linrunner/TLP #205: [enhancement] intel CPU governor

mattia-b89, 2016-05-16:
better explain the intel_pstate section, since default is `ondemand` but intel_pstate doesn't support it!

linrunner, 2016-05-16:
Feel free to suggest a better text or send a pull request!

mattia-b89, 2016-05-16:
A point is to add the official kernel page:
https://www.kernel.org/doc/Documentation/cpu-freq/intel-pstate.txt

linrunner, 2016-05-19:
* default changed --> https://github.com/linrunner/TLP/commit/531b17e53888b66c0f650ccd2377c88102556f41
* Config page changed --> http://linrunner.de/en/tlp/docs/tlp-configuration.html#scaling
* tlp-stat -p: show available scaling govs  --> https://github.com/linrunner/TLP/commit/a6121c2e0a577e4c353d1b4b2404fde6712474bf

OK for you?

> A point is to add the official kernel page:
Is already linked on the config page.

closed 2016-05-22 by linrunner