linrunner/TLP #207: [future enhancement] schedutil governor

mattia-b89, 2016-05-22:
...linrunner, don't blame me...
I'd like to open this bug as "reminder"

kernel 4.7 will introduce new governor called `schedutil`
in order to be able to handle this new feature, we have to add it and maybe(?) update defaults

linrunner, 2016-05-22:
Your contributions would be much more welcome (and less precocious) if you do some homework in advance:

* giving references for the related kernel change
* suggesting a concrete implementation path
* prepare patches / pull requests

The issue tracker is not intended as a "make a wish" tool.

mattia-b89, 2016-05-22:
I perfectly understand you but unfortunately I can contribute in this way
only. I thank you for your previous work and time you devote!

I haven't provided any doc since it is early... at least official kernel
documentation page is empty. Let's wait official 4.7 release.

Anyway work to do is adding the new 'value' in the governor array and
mention it in the guide but this is only an 'user' point-of-view, I haven't
read the entire code... probably this new governor will not accept some
values accepted instead by other governor, so we need an IF construct... a
further because precocious development is to edit default with this new
governor if it worth (
http://www.phoronix.com/scan.php?page=article&item=linux-47-schedutil&num=1)

linrunner, 2016-08-28:
Settings docs updated.

closed 2016-08-28 by linrunner