linrunner/TLP #210: Fedora repository GPG key concerns

jkpl, 2016-05-23:
Hi, thanks for the awesome tools! I'm concerned about the GPG key that you use for the Fedora repository. First of all, the appears to be expired:

```
$ curl -sL 'http://repo.linrunner.de/fedora/tlp/repos/RPM-GPG-KEY-tlp' | gpg --with-fingerprint
pub  dsa1024/C2FB431C 2012-04-16 [expires: 2013-04-16]
      Key fingerprint = 8AD5 9D64 3341 C382 1333  3FBD C89A 1C1A C2FB 431C
uid                   Andreas Roederer (tlp Repository) <tlp@warpnine.de>
sub  elg1024/00C9A31B 2012-04-16 [expires: 2013-04-16]
```

Second, I could not find the key on any key servers I looked.

```
$ gpg --keyserver hkp://pool.sks-keyservers.net --search-key 'tlp@warpnine.de'
gpg: error searching keyserver: No data
gpg: keyserver search failed: No data
```

Also, I could not find any mention of the key or its fingerprint anywhere outside the repository.

linrunner, 2016-05-25:
Hi,

thanks for you hint. The Fedora repo maintainer will look into that.

linrunner, 2016-05-26:
1) Key is no longer expired, please re-check.
2) No solution yet.

jkpl, 2016-05-27:
Verified that key the key is no longer expired.

```
$ curl -sL 'http://repo.linrunner.de/fedora/tlp/repos/RPM-GPG-KEY-tlp' | gpg2 --with-fingerprint
pub  1024D/C2FB431C 2012-04-16 [expires: 2017-07-13]
      Key fingerprint = 8AD5 9D64 3341 C382 1333  3FBD C89A 1C1A C2FB 431C
uid                            Andreas Roederer (tlp Repository) <tlp@warpnine.de>
sub  1024g/00C9A31B 2012-04-16 [expires: 2017-07-13]
```

stefangweichinger, 2017-08-20:
I also have problems updating tlp-packages with current Fedora 26.

``` Warnung: /var/cache/dnf/tlp-fc68e1be389b3a64/packages/akmod-tp_smapi-0.42-3.fc26.x86_64.rpm: Header V4 RSA/SHA256 Signature, Schlüssel-ID 1482d93f: NOKEY
Importing GPG key 0xC2FB431C:
 Userid  Warnung: /var/cache/dnf/tlp-fc68e1be389b3a64/packages/akmod-tp_smapi-0.42-3.fc26.x86_64.rpm: Header V4 RSA/SHA256 Signature, Schlüssel-ID 1482d93f: NOKEY
Importing GPG key 0xC2FB431C:
 Userid     : "Andreas Roederer (tlp Repository) <tlp@warpnine.de>"
 Fingerprint: 8AD5 9D64 3341 C382 1333 3FBD C89A 1C1A C2FB 431C
 From       : /etc/pki/rpm-gpg/RPM-GPG-KEY-tlp
Ist dies in Ordnung? [j/N]:j
Schlüssel erfolgreich importiert
Importieren der Schlüssel hat nicht geholfen, falsche Schlüssel?
Importieren der Schlüssel hat nicht geholfen, falsche Schlüssel?

Öffentlicher Schlüssel für akmod-tp_smapi-0.42-3.fc26.x86_64.rpm ist nicht installiertFehlgeschlagenes Paket ist: akmod-tp_smapi-0.42-3.fc26.x86_64
 GPG-Schlüssel sind eingerichtet als: file:///etc/pki/rpm-gpg/RPM-GPG-KEY-tlp

Öffentlicher Schlüssel für tp_smapi-0.42-3.fc26.noarch.rpm ist nicht installiertFehlgeschlagenes Paket ist: tp_smapi-0.42-3.fc26.noarch
 GPG-Schlüssel sind eingerichtet als: file:///etc/pki/rpm-gpg/RPM-GPG-KEY-tlp   : "Andreas Roederer (tlp Repository) <tlp@warpnine.de>"
 Fingerprint: 8AD5 9D64 3341 C382 1333 3FBD C89A 1C1A C2FB 431C
 From       : /etc/pki/rpm-gpg/RPM-GPG-KEY-tlp
Ist dies in Ordnung? [j/N]:j
Schlüssel erfolgreich importiert
Importieren der Schlüssel hat nicht geholfen, falsche Schlüssel?
Importieren der Schlüssel hat nicht geholfen, falsche Schlüssel?

Öffentlicher Schlüssel für akmod-tp_smapi-0.42-3.fc26.x86_64.rpm ist nicht installiertFehlgeschlagenes Paket ist: akmod-tp_smapi-0.42-3.fc26.x86_64
 GPG-Schlüssel sind eingerichtet als: file:///etc/pki/rpm-gpg/RPM-GPG-KEY-tlp

Öffentlicher Schlüssel für tp_smapi-0.42-3.fc26.noarch.rpm ist nicht installiertFehlgeschlagenes Paket ist: tp_smapi-0.42-3.fc26.noarch
 GPG-Schlüssel sind eingerichtet als: file:///etc/pki/rpm-gpg/RPM-GPG-KEY-tlp

runcom, 2017-08-21:
Fedora 26 failing indeed:
```
warning: /var/cache/dnf/tlp-fc68e1be389b3a64/packages/tlp-release-1.0.8-1.fc26.noarch.rpm: Header V4 RSA/SHA256 Signature, key ID 1482d93f: NOKEY
Importing GPG key 0xC2FB431C:
 Userid     : "Andreas Roederer (tlp Repository) <tlp@warpnine.de>"
 Fingerprint: 8AD5 9D64 3341 C382 1333 3FBD C89A 1C1A C2FB 431C
 From       : /etc/pki/rpm-gpg/RPM-GPG-KEY-tlp
Is this ok [y/N]: y
Key imported successfully
Import of key(s) didn't help, wrong key(s)?

Public key for tlp-release-1.0.8-1.fc26.noarch.rpm is not installedFailing package is: tlp-release-1.0.8-1.fc26.noarch
 GPG Keys are configured as: file:///etc/pki/rpm-gpg/RPM-GPG-KEY-tlp

The downloaded packages were saved in cache until the next successful transaction.
You can remove cached packages by executing 'dnf clean packages'.
Error: GPG check FAILED
```

sabotatore, 2017-08-21:
I have the same issue on Fedora 26.

cunio, 2017-08-21:
I'm confirming last issues on Fedora 26

Thorashh, 2017-08-21:
You can safely import the right key manually.
`rpm --import /etc/pki/rpm-gpg/RPM-GPG-KEY-tlp-fedora-26-primary`

After that, please check your /etc/yum.repos.d/tlp*.repo files. You are using the old ones. 

Explanation. I have replaced the old signature key with a new one.

OR use:
`dnf remove tlp-release`
`dnf install http://repo.linrunner.de/fedora/tlp/repos/releases/tlp-release.fc26.noarch.rpm`

stefangweichinger, 2017-08-21:
2nd suggestion worked for me right now, thanks. Will repeat on other systems asap.

linrunner, 2017-08-22:
Right. For F26 you have to install a new repo key. I added this to [Installation/Fedora](http://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html#fedora).

closed 2017-08-22 by linrunner

andypillip, 2018-07-06:
I'm on F28 and getting the warning about the key, which seems to be a new one now:

> Warnung: /var/cache/dnf/tlp-65642a015776da19/packages/akmod-tp_smapi-0.43-1.fc28.x86_64.rpm: Header V4 RSA/SHA256 Signature, Schlüssel-ID 1a853c73: NOKEY
> Importing GPG key 0x1A853C73:
>  Userid     : "TLP Repository for Fedora (28) <tlp@warpnine.de>"
>  Fingerprint: C807 AEB6 3DD0 4587 E695 9DD2 455A 80BA 1A85 3C73
>  From       : /etc/pki/rpm-gpg/RPM-GPG-KEY-tlp-28

I don't find the fingerprint anywhere.

inhumantsar, 2018-07-11:
@andypillip I ran into this issue on FC28 as well. Running the remove/install again worked:

```
dnf remove tlp-release
dnf install http://repo.linrunner.de/fedora/tlp/repos/releases/tlp-release.fc28.noarch.rpm
```

linrunner, 2018-08-02:
Signature fingerprints are documented here now: https://linrunner.de/en/tlp/docs/tlp-fedora-repo-keys.html

andypillip, 2020-10-29:
Hi again. Would you mind having a look at this fingerprint mismatch issue?

https://www.linrunner.de/tlp/installation/fedora.html
lists the fingerprint for Fedora 33 as follows
> 1E4F 2F53 A348 6025 FC4E FD86 7704 0BAF FA30 D1C8

If I follow the instructions above on that page, and when updating to Fedora 33, I'm getting the following:
>  0D3C F36C EB28 B582 D6DE F296 82B6 D96F 1381 B02A

Why is that? Thanks!

Thorashh, 2020-10-30:
Moin

Thank you for the info.

"0D3C F36C EB28 B582 D6DE F296 82B6 D96F 1381 B02A" is the correct Fingerprint for the **F33** Key.

"1E4F 2F53 A348 6025 FC4E FD86 7704 0BAF FA30 D1C8" is the correct Fingerprint for the **F34** Key.

Update to the doc follows soon