linrunner/TLP #211: ASPM settings

mattia-b89, 2016-05-26:
even if `tlp` default handle ASPM, thanks to:
`PCIE_ASPM_ON_AC=performance`
`PCIE_ASPM_ON_BAT=powersave`

settings are not applied:
`[mattia@arch-xps ~]$ cat /sys/module/pcie_aspm/parameters/policy`
`[default] performance powersave`

does this issue rely to default kernel behaviour to disable this feature?
`[mattia@arch-xps ~]$ dmesg | grep aspm`
`...`
`[    0.090343] ACPI FADT declares the system doesn't support PCIe ASPM, so disable it`
`...`

if it does, we have to explain the situation (that is (the idea of mine): ASPM is disabled by default if BIOS doesn't declare its support; you can check it in dmesg; AT-YOUR-OWN-RISK you can enable it with this kernel parameter `pcie_aspm=force`) on the guide and configuration comments

FadeMind, 2016-05-26:
@mattia-b89 ASPM can be forcibly power on via `pcie_aspm=force` and then TLP can handle it .See `sudo tlp-stat`:
[image]

mattia-b89, 2016-05-26:
@FadeMind it's exactly what I've just written...!!!

linrunner, 2016-05-26:
No, i won't tell users to make use of `pcie_aspm=force`. It has little (if any) benefit but great potential to make the system unstable. Those who absolutely want to try, will find it anyway.

closed 2016-05-26 by linrunner

mattia-b89, 2016-05-27:
Absolutely, I don't want to do it, too!
But I'd and only like to find a mention about it... that is only what I've
written in the last part of the OP