linrunner/TLP #214: Option to "fullcharge" until AC power disconnected

frangio, 2016-06-04:
Currently the fullcharge settings will hold until the next reboot, but I would find it more useful if the thresholds went back to normal as soon as AC is disconnected.

frangio, 2016-07-15:
I'm thinking about implementing this by creating a subcommand `fullchargeonce` (or something like that) that does the same thing `fullcharge` does and also launches a background process that watches for an AC unplug event and then restores the thresholds and exits. I haven't yet looked into it much, but I know I can get such an event from the UPower service on the system message bus. Does this sound okay?

frangio, 2016-07-15:
Another option is using `udevadm monitor`.

frangio, 2016-07-19:
I've implemented this functionality on the [`setchargewhileac`](https://github.com/frangio/TLP/tree/setchargewhileac) branch on my fork. I ended up using the existing udev rule which executes `tlp auto` and just adding the necessary code to this subcommand.

If you're interested in including this feature I'd be happy to hear your comments on the implementation and I'll make the necessary changes before doing a pull request.

linrunner, 2016-07-20:
Hi, thank you. I'll look into your code (after the 0.9 release).

frangio, 2016-08-09:
After using it for a while I realized that the "setchargewhileac" command was kind of an over the top solution and not really comfortable to use.

I've implemented a much simpler alternative on the [`restorethreshonbat`](https://github.com/frangio/TLP/tree/restorethreshonbat) branch. It consists of a setting to determine if the default thresholds should be restored whenever the power source changes to battery. It's disabled by default. It's just three lines of code and an extra configuration setting.

linrunner, 2016-09-10:
Simple yet elegant solution. Done it myself --> https://github.com/linrunner/TLP/commit/af507296205cd1ecaf8fe3b2474a99ae99ba7e4a

Thank you.

closed 2016-09-10 by linrunner