linrunner/TLP #215: computer reboot instead of power off

zz77zz, 2016-06-06:
When I install TLP on my system it reboots (after 2-3 sec) instead of powering off.
I removed (with purge) TLP and the problem was solved. It went back when I installed again TLP...
I even tried a completely new installation of the OS because I was afraid of having set up the mess inside. I am using a brand new HP Zbook 15 (so I'm waiting to solve this issue before pursuing system config).
Here is the output of "tlp-stat":

--- TLP 0.8 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
TLP_DEFAULT_MODE=AC
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau"
USB_AUTOSUSPEND=1
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0

+++ System Info
System         = Hewlett-Packard A3008CD10003 HP ZBook 15 G2
BIOS           = M70 Ver. 01.14
Release        = BunsenLabs GNU/Linux 8.5 (Hydrogen)
Kernel         = 4.5.0-0.bpo.2-amd64 #1 SMP Debian 4.5.4-1~bpo8+1 (2016-05-13) x86_64
/proc/cmdline  = BOOT_IMAGE=/vmlinuz-4.5.0-0.bpo.2-amd64 root=/dev/mapper/debian--vg-root ro initrd=/install/initrd.gz quiet
Init system    = systemd

+++ System Status
TLP power save = enabled
power source   = AC

+++ Processor
CPU Model      = Intel(R) Core(TM) i7-4710MQ CPU @ 2.50GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   800000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  3500000 [kHz]

/sys/devices/system/cpu/cpu1/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor  = powersave
[... 122 lines cut ...]
+++ USB
autosuspend        = enabled
device whitelist   = (not configured)
device blacklist   = (not configured)
wwan blacklist     = enabled

Bus 002 Device 002 ID 8087:8000 control = auto, autosuspend_delay_ms =     0 -- Intel Corp.  (hub)
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 001 Device 002 ID 8087:8008 control = auto, autosuspend_delay_ms =     0 -- Intel Corp.  (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 004 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 003 Device 003 ID 04ca:704d control = auto, autosuspend_delay_ms =  2000 -- Lite-On Technology Corp.  (uvcvideo)
Bus 003 Device 002 ID 138a:003f control = auto, autosuspend_delay_ms =  2000 -- Validity Sensors, Inc.  (no driver)
Bus 003 Device 004 ID 8087:07dc control = auto, autosuspend_delay_ms =  2000 -- Intel Corp.  (btusb)
Bus 003 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)

+++ Battery Status
/sys/class/power_supply/BAT0/manufacturer                   = Hewlett-Packard
/sys/class/power_supply/BAT0/model_name                     = Primary
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/charge_full_design             =   5000 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   5000 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   5000 [mAh]
/sys/class/power_supply/BAT0/current_now                    =      0 [mA]
/sys/class/power_supply/BAT0/status                         = Full

linrunner, 2016-06-07:
Hi, does this happen on AC or battery or both?

zz77zz, 2016-06-07:
Hello, it happens both on AC and battery.

linrunner, 2016-06-07:
First try is to disable Runtime PM completely by commenting the related lines in the config:

     #RUNTIME_PM_ON_AC=on
     #RUNTIME_PM_ON_BAT=auto

And reboot of course.

zz77zz, 2016-06-11:
Hi, the problem disappeared when I manually disabled Wake-on-LAN in the BIOS (this option is well hidden in this HP BIOS!). I know this solution has been proposed several times in forums, I'm sorry I opened an issue for that... Anyway, it seems something was related with TLP because there was no problem without it installed. 
For information, disabling Runtime PM had no effect.
Thank you.

linrunner, 2016-06-11:
Thanks for your feedback.

closed 2016-06-11 by linrunner