linrunner/TLP #216: Note: Xfce Power Manager  reports Charging  when it actually isn't!

ActualizeInMaterial, 2016-06-07:
This is not an issue with tlp, just a note. This threw me off a little because I was worried that the batt thresholds weren't working! (on my T400)

First, the reported percent in `Xfce Power Manager` is 68% (1 less than what tlp reports - 69% )
Second, `Xfce Power Manager` reports `Charging` when it actually isn't! (as tlp reports correctly that it isn't!)

```
# xfce4-power-manager --version

Xfce Power Manager 1.6.0

Part of the Xfce Goodies Project
http://goodies.xfce.org

Licensed under the GNU GPL.

# tlp-stat|grep BAT0
START_CHARGE_THRESH_BAT0=40
STOP_CHARGE_THRESH_BAT0=50
+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/devices/platform/smapi/BAT0/manufacturer               = LGC
/sys/devices/platform/smapi/BAT0/model                      = 42T4653
/sys/devices/platform/smapi/BAT0/manufacture_date           = 2011-07-25
/sys/devices/platform/smapi/BAT0/first_use_date             = 2011-12-16
/sys/devices/platform/smapi/BAT0/cycle_count                =    117
/sys/devices/platform/smapi/BAT0/design_capacity            =  56160 [mWh]
/sys/devices/platform/smapi/BAT0/last_full_capacity         =  60740 [mWh]
/sys/devices/platform/smapi/BAT0/remaining_capacity         =  41610 [mWh]
/sys/devices/platform/smapi/BAT0/remaining_percent          =     69 [%]
/sys/devices/platform/smapi/BAT0/remaining_running_time_now = not_discharging [min]
/sys/devices/platform/smapi/BAT0/remaining_charging_time    = not_charging [min]
/sys/devices/platform/smapi/BAT0/power_now                  =      0 [mW]
/sys/devices/platform/smapi/BAT0/power_avg                  =      0 [mW]
/sys/devices/platform/smapi/BAT0/state                      = idle
/sys/devices/platform/smapi/BAT0/start_charge_thresh        =     40 [%]
/sys/devices/platform/smapi/BAT0/stop_charge_thresh         =     50 [%]
/sys/devices/platform/smapi/BAT0/force_discharge            =      0

```
(those are the same value after 10 minutes of waiting for them to potentially change - they didn't! in case anyone was wondering at this point)

[image]

[image]

closed 2016-06-07 by ActualizeInMaterial

linrunner, 2016-06-07:
The TLP issue tracker is for bug reports only, not for mere support questions or "notes".  

Panel applets showing incorrect charging status is mentioned in the [FAQ](http://linrunner.de/en/tlp/docs/tlp-faq.html#battery): "Why does the panel applet show the battery state "charging" despite charge thresholds are effective?". 

Please carefully check the FAQ before opening an issue. Thanks.

ActualizeInMaterial, 2016-06-11:
Thanks!

TLP rounds up* the percent value, which is why I was seeing the difference.

\* that is, truncate the fraction and(if fraction is non-zero? perhaps, only then) add 1. (didn't check the code, so I'm only guessing from some value samples)

```
4161000/60740
68.50510372077708264734
```

```
/sys/devices/platform/smapi/BAT0/design_capacity            =  56160 [mWh]
/sys/devices/platform/smapi/BAT0/last_full_capacity         =  60740 [mWh]
/sys/devices/platform/smapi/BAT0/remaining_capacity         =  40130 [mWh]
/sys/devices/platform/smapi/BAT0/remaining_percent          =     67 [%]
```

```
4013000/60740
66.06848864010536713862
```

So 66.06% became 67% in TLP.

Basically, TLP reports a higher percentage (by 1%).
Not that it matters, but that explains why I see the difference between it and xfce power manager.