linrunner/TLP #218: akmod-acpi_call not available for Fedora 24

kc1212, 2016-06-25:
I'm doing `dnf install tlp akmod-acpi_call` but I'm getting the error `No package akmod-acpi_call available.` from `dnf`. This is with the TLP repo. 

Is `akmod-acpi_call` available for Fedora 24 yet?

linrunner, 2016-06-26:
There is no package for F24 yet because akmod prerequisites from RPMFusion are missing:

- Packages available as updates-testing only 
- Packages not signed (e.g. buildsys-build-rpmfusion)

We have to wait until the RPMFusion project has sorted this. I suggest you complain there.

kc1212, 2016-06-26:
On rpmfusion, updates and release repos are available from looking at
their website. I am able to find akmods in rpmfusion-free on my machine,
I'm guessing that's one of the prerequisites.

linrunner, 2016-06-27:
Packages `akmod-acpi_call` and `akmod-tp_smapi` (0.42) are available now (via `--enablerepo tlp-updates-testing`). Please check and report.

kc1212, 2016-06-27:
It's working now, thank you for the great software!

closed 2016-06-27 by kc1212