linrunner/TLP #219: Installation instructions for Fedora may need updating (systemd-rfkill)

Favorablestream, 2016-06-27:
Documentation I used:
http://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html#fedora

The instructions reccomend diabling systemd's rfkill service:

systemctl mask systemd-rfkill@.service

On my system (Fedora 24) I had to use the alternate name instead:

systemctl mask systemd-rfkill

The [systemd man page](http://man7.org/linux/man-pages/man8/systemd-rfkill@.service.8.html) lists both names but only the second one works for me.

Thanks

linrunner, 2016-06-27:
[Done](http://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html#fedora). Thanks for your hint.

closed 2016-06-27 by linrunner

GabVenturato, 2016-11-11:
Hi, talking about Fedora 24:
doing `tlp-stat -s` at the end I see this message:

> Notice: systemd-rfkill@.service is not masked -- invoke "systemctl mask systemd-rfkill@.service" to correct this!

I did already `systemctl mask systemd-rfkill`, but the notice is still there.
After searching a bit what I can see is that if I do `$ systemctl list-unit-files | grep rfkill` the output is:
`systemd-rfkill.service                      masked`
`systemd-rfkill@.service                     static`
`systemd-rfkill.socket                       static`

the first one is masked because of the command suggested in documentation, but the others are not masked.
the question now is: what is needed to be masked?
and if I run `systemctl mask systemd-rfkill@.service` it says 

> Failed to execute operation: Invalid argument

at15, 2017-01-08:
I am using Fedora 25 and have the same problem as @GabVenturato I tried #166 to mask the socket (though it is not mentioned in the Notice) and as for `systemd-rfkill@.service` I got the following error 

> Failed to get properties: Unit name systemd-rfkill@.service is missing the instance name.

running `systemctl list-unit-files | grep rfkill` I have 

````
systemd-rfkill.service                      masked   
systemd-rfkill@.service                     static   
systemd-rfkill.socket                       masked   
````

linrunner, 2017-01-08:
Remove the file /etc/systemd/system/systemd-rfkill@.service – i guess it's some kind of artifact from previous Fedora versions.

Don't mess with systemd-rfkill.socket please, masking it is not necessary.

at15, 2017-01-08:
@linrunner it is a one line file, though I have a fresh Fedora 25 install, thanks for your help :smile: 
````
at15@acer:/etc/systemd/system|⇒  cat systemd-rfkill@.service
#
````

ir193, 2017-01-08:
Actullay I got:

```
rpm -qf /etc/systemd/system/systemd-rfkill@.service
tlp-0.9-1.fc25.noarch
```

tlp installed from fedora repo

at15, 2017-01-08:
Found on https://apps.fedoraproject.org/packages/tlp/sources/ L64-66

````
#THIS IS A WORKAROUND FOR RFKILL!
ln -sf /dev/null systemd-rfkill@.service
install -D systemd-rfkill@.service %{buildroot}%{_sysconfdir}/systemd/system/systemd-rfkill@.service
````

@linrunner

linrunner, 2017-01-08:
I see. Just filing a bug against the package.

Mystro256, 2017-01-16:
I've pushed the fix to fedora testing and rawhide; it'll land in the next few days. There shouldn't be any need to mask the systemd service starting with tlp-0.9-2, as it will do it automatically after package upgrade.

linrunner, 2017-01-26:
F24/F25 packages are in main repo by now.

Mystro256, 2017-01-31:
FYI:

For those who use EPEL7, I've just submitted the fix to the epel7 stable repo; it should land in a day or two. This issue does not affect epel6.