linrunner/TLP #220: Refresh Rate

Ablablab, 2016-08-04:
My laptop is an HP Elitebook Folio with a hd-ready led display, it's a professional laptop and battery life was one of the biggest target of HP. I noticed that in Windows, preinstalled drivers switch refresh rate of display to 40Hz in battery mode to 60Hz in AC mode. 

I made some researches. Only on CRT display the refresh rate has a significative impact in battery life, in fact on lcd refresh rate is not significant in power consumption, in led displays has no impact.
On the other hand, set refresh rate to 40Hz instead of 60Hz, has a large impact on graphic rendering, are 2/3 number of renderized frames. 

Now I've used xrandr -r 40 or 60 when the laptop switch to battery mode or ac mode and i'm glad of this mod. 

Maybe it could be a next feature? i thingk that tonight or later i can fork your project and add this little mod, so you can approve or not. 

I can't see the difference, even with high or low brightness, between 40 and 60Hz.

What are you thinking about?

linrunner, 2016-08-11:
Hi, i'll check your suggestion after the 0.9 release.

linrunner, 2016-09-12:
I just tried your suggested command with three ThinkPads at hand: X200, X220, T450s.

Regardless which rate i try – including the ones xrandr -q shows for the actual resolution – xrandr seems to ignore all of them silently. I can see no change. Add the fact that the 2 newer models modes all advertise only values near 60 Hz for their native resolution.

When comparing power consumption between 60 and 40 Hz there is no noticable change. 

Just now i'm far from being convinced to add this feature.

Did you measure the difference of power consumption for your hardware?

closed 2016-10-09 by linrunner

ilovelinux, 2026-08-29:
This feature request has been rejected. Please see https://github.com/linrunner/TLP/issues/467#issuecomment-583977490.

<sub>Note: I searched for this feature on Google and this was the first result. I'm linking the rationale of rejection for people running into this issue</sub>