linrunner/TLP #222: Wired network not working after resuming from suspend (Ubuntu 16.04)

hotice, 2016-08-23:
When TLP is installed, the weird network on my laptop does not work after resuming from suspend. Tested in Ubuntu 16.04 with TLP 0.8 and 0.9. The issue can be replicated 100% of the time - it always occurs with TLP, never when TLP is not installed.

I tried all the suggestions mentioned in this bug report: https://github.com/linrunner/TLP/issues/181 , with no success.

Here's the output of tlp-stat (ignore the battery being pretty much dead :D): https://gist.github.com/hotice/ed58c1215c4ead10a561f28f781b8c9c

linrunner, 2016-08-23:
Hi,

at the end of the day it's always a faulty kernel powersave feature that TLP exposes by activating it ... honestly ;-)

1. Happens on battery or AC or both?
2. Try disabling Runtime PM completely by commenting out both lines:

    #RUNTIME_PM_ON_AC=on
    #RUNTIME_PM_ON_BAT=auto 

3. Try WOL_DISABLE=Y 
4. Try to restart NetworkManager after resume

    sudo systemctl restart NetworkManager

For 2. and 3. make shure to poweroff before testing the new setting

hotice, 2016-08-23:
It only happens on AC. I tried all your solutions and one worked: setting ``WOL_DISABLE=N`` in the config. Thanks!

vooze, 2016-08-23:
Seems to work for me, can't reproduce with tlp 0.9. (Tested with AC + Cable + suspend)

linrunner, 2016-08-26:
Added to the [FAQ](http://linrunner.de/en/tlp/docs/tlp-faq.html#ethresume) for good measure :-)

@vooze: i can't reproduce it either ;-).

closed 2016-08-28 by linrunner