linrunner/TLP #223: Wrong power source

dzanotto, 2016-08-26:
If I power on my laptop on battery and then connect it to AC, the power source reported by tlp-stat is battery. Apparently the switch between battery-AC and AC-battery works well if the laptop is on AC when powered on.

TLP 0.9

System info:
System = Dell Inc. XPS 15 9550
BIOS = 01.02.00
Release = Ubuntu 16.04.1 LTS
Kernel = 4.4.0-36-generic #55-Ubuntu SMP Thu Aug 11 18:01:55 UTC 2016 x86_64
/proc/cmdline = BOOT_IMAGE=/vmlinuz-4.4.0-36-generic.efi.signed root=/dev/mapper/ubuntu--vg-root ro quiet splash vt.handoff=7
Init system = systemd v229
Boot mode = UEFI

dzanotto, 2016-08-26:
What information can I add to make the issue more understandable?

linrunner, 2016-08-26:
Hi,

nice, you left out the interesting part of tlp-stat e.g. the "TLP Status" section ;-).

So here we go. I need the full output of

    sudo tlp-stat
    grep '.*' /sys/class/power_supply/*/{online,type}

on **battery and AC** via https://gist.github.com/ please.

dzanotto, 2016-08-26:
You can find the requested data [here](https://gist.github.com/anonymous/ad4c9a4550deba8f177b7914139c4aa7)
*_battery.txt -> data on battery
*_ac.txt -> data on ac (connected after boot)
*_acok.txt -> data on ac (connected before boot)
Apparently when the laptop boot on battery, /sys/class/power_supply/AC/ doesn't exists.

linrunner, 2016-08-28:
> Apparently when the laptop boot on battery, /sys/class/power_supply/AC/ doesn't exists.

That's the point: if it isn't there, TLP can't detect AC. Currently i see no way to work around that in TLP.

Looks like a BIOS bug in the [ACPI DSDT](https://wiki.archlinux.org/index.php/DSDT) or some problem in the interaction between ACPI and kernel. You could try:
* BIOS update, 01.02.13 is available on the [Dell site](http://www.dell.com/support/home/us/en/04/product-support/product/xps-15-9550-laptop/drivers?rvps=y)
* [Mainline kernel ](http://kernel.ubuntu.com/~kernel-ppa/mainline/) e.g. 4.7 or 4.8rcX
* Check and fix the DSDT yourself (see link above)
* File your complaint at [Dell](http://en.community.dell.com/techcenter/os-applications/f)

Just to have the symptoms complete, please boot on battery once more and show the output of

    sudo tlp-stat --pev

after plugging in AC.

dzanotto, 2016-08-30:
I apologize for the delayed response.
You can find the output of tlp-stat --pev [here](https://gist.github.com/dzanotto/799eceb13e0d8346166fdc65c366c4fe).
I'll try your other hints ASAP.

dzanotto, 2016-09-07:
Quick update:
- the BIOS update didn't fix it
- with newer kernel images I'm having some other troubles, so I can't go this way
- before trying to fix the DSDT, I've filed a [kernel bug](https://bugzilla.kernel.org/show_bug.cgi?id=156171)

linrunner, 2016-09-10:
The kernel dev suggests to complain to Dell too: may i close here?

dzanotto, 2016-09-11:
Of course, thanks for your help

closed 2016-09-11 by dzanotto