linrunner/TLP #224: ThinkPad X1 Carbon (4th gen, 2016) - Missing tp_smapi and acpi_call kernel modules

leodido, 2016-08-30:
I have followed your instructions at http://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html.

Nevertheless I'm obtaining the following output from `tlp-stat`.

```
+++ ThinkPad Extended Battery Functions
tp-smapi   = inactive (kernel module 'tp_smapi' not installed)
tpacpi-bat = inactive (kernel module 'acpi_call' not installed)
```

Furthermore `modprove -v` states that it does not find the modules in the kernel modules' directory.

Any hints or suggest?

linrunner, 2016-08-30:
Hi, it doesn't matter, because the X1 Carbon 4th Gen. does not support tp-smapi anyway. You may uninstall the package.

Hint: TLP 0.9 (available in the F24 testing repo) will tell you exactly this, see the [FAQ](http://linrunner.de/en/tlp/docs/tlp-faq.html#kernmod) for details.

leodido, 2016-08-31:
Hi @linrunner, ok. Perfect.

And thanks for your wonderful piece of software.

linrunner, 2016-08-31:
You're welcome.

closed 2016-08-31 by linrunner