linrunner/TLP #225: ERROR: systemd-udevd[1278]: Process '/usr/bin/tlp auto' failed with exit code 4.

halocaridina, 2016-09-02:
Hi linrunner,

Just started noticing the above error in my journal upon changes in power state. First instance is unplugging AC, second is returning to AC a few seconds later:

`Sep 02 16:44:00 wopr systemd-udevd[1278]: Process '/usr/bin/tlp auto' failed with exit code 4.`
`Sep 02 16:44:08 wopr systemd-udevd[1378]: Process '/usr/bin/tlp auto' failed with exit code 4.`

**For diagnositics, output of "tlp stat" on AC on update-to-date Arch laptop is:**

--- TLP 0.9 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
TLP_DEFAULT_MODE=AC
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
CPU_SCALING_GOVERNOR_ON_AC=performance
CPU_SCALING_GOVERNOR_ON_BAT=powersave
CPU_MIN_PERF_ON_AC=0
CPU_MAX_PERF_ON_AC=100
CPU_MIN_PERF_ON_BAT=0
CPU_MAX_PERF_ON_BAT=50
CPU_BOOST_ON_AC=1
CPU_BOOST_ON_BAT=1
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="ata-TOSHIBA_THNSNJ256GCSU_552S1043TB4W ata-Crucial_CT250MX200SSD6_162112C6D4EF"
DISK_APM_LEVEL_ON_AC="255 255"
DISK_APM_LEVEL_ON_BAT="255 255"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
AHCI_RUNTIME_PM_TIMEOUT=15
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=off
WIFI_PWR_ON_BAT=on
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau"
USB_AUTOSUSPEND=1
USB_BLACKLIST_WWAN=1
USB_AUTOSUSPEND_DISABLE_ON_SHUTDOWN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0
DEVICES_TO_DISABLE_ON_LAN_CONNECT="wifi"
DEVICES_TO_ENABLE_ON_LAN_DISCONNECT="wifi"

+++ System Info
System         = Dell Inc. Latitude 3150
BIOS           = A05
Release        = "Arch Linux"
Kernel         = 4.7.2-2-ck #1 SMP PREEMPT Fri Sep 2 13:44:48 EDT 2016 x86_64
/proc/cmdline  = initrd=\intel-ucode.img initrd=\initramfs-linux-ck.img root=PARTUUID=c47b0f9f-23fd-47bc-ada2-7aaca8ef916e pcie_aspm=force acpi_osi=Linux acpi_osi='!Windows 2012' resume=PARTUUID=c47b0f9f-23fd-47bc-ada2-7aaca8ef916e resume_offset=51200 ipv6.disable_ipv6=1 rw ####usbcore.autosuspend=15 rw
[... 431 lines cut ...]
/sys/class/power_supply/hid-00:1f:20:e1:13:39-battery/manufacturer = (not available)
/sys/class/power_supply/hid-00:1f:20:e1:13:39-battery/model_name = Bluetooth Mouse M557
/sys/class/power_supply/hid-00:1f:20:e1:13:39-battery/cycle_count = (not supported)
/sys/class/power_supply/hid-00:1f:20:e1:13:39-battery/status = Discharging

Charge                                                      =   97.2 [%]
Capacity                                                     =  100.0 [%]

Running "sudo tlp ac" or "sudo tlp bat" produces no errors on the commandline nor entries in the journal. 

Could this error be associated with the recent update to dbus:

`[2016-08-26 12:42] [ALPM] removed python-dbus (1.2.4-1)`
`[2016-08-26 22:08] [ALPM] installed python-dbus (1.2.4-1)`
`[2016-08-28 00:37] [ALPM] removed libdbus (1.10.8-1)`
`[2016-08-28 00:37] [ALPM] upgraded dbus (1.10.8-1 -> 1.10.10-3)`

which has been causing various issues:

https://bbs.archlinux.org/viewtopic.php?id=216001
https://bbs.archlinux.org/viewtopic.php?id=216523

Thanks in advance and please let me know if you need additional info.

Halocaridina

linrunner, 2016-09-04:
Hi, 

short answer: works as designed. Ignore the "errors".

Long answer: rc = 4 means that an instance of tlp is already running applying settings. E.g. tlp is "locked" and the new instance just terminates without doing anything. This happens because on most laptops a change of power source triggers multiple udev events. TLP < 0.9 didn't block these duplicate invocations.

halocaridina, 2016-09-04:
Hi linrunner,

Ok, good to know that this really isn't an "issue".  This is resolved and can be closed.

Cheers,

Halocaridina

closed 2016-09-04 by halocaridina