linrunner/TLP #226: Fan high speed never goes down - TLP 0.9 on Thinkpad P50 with Ubuntu 16.04.1

greengene, 2016-09-06:
I have installed TLP 0.9 on Thinkpad P50 with Ubuntu 16.04.1 LTS.

If I don't do anything that requires too much of the CPU, it works fine.

But once I do use the CPU, let's say when I restart Firefox: the fan starts running at high speed, about 4000 RPM - and it never stops.

If I disable TLP and do the same thing (i.e. restarting Firefox), the fan will run high for 1-2 seconds and then go back to a low normal speed, about 2300 RPM, which is what it should do.

How can I solve this?

`+++ System Info`
`System         = LENOVO ThinkPad P50 20ENCTO1WW`
`BIOS           = N1EET52W (1.26 )`
`Release        = Ubuntu 16.04.1 LTS`
`Kernel         = 4.4.0-34-generic #53-Ubuntu SMP Wed Jul 27 16:06:39 UTC 2016 x86_64`
`/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-4.4.0-34-generic.efi.signed root=UUID=8f91a113-954c-48cd-a597-122fdb459100 ro quiet splash vt.handoff=7`
`Init system    = systemd v229`
`Boot mode      = UEFI`

`+++ TLP Status`
`State          = enabled`
`Last run       = 04:56:58 PM,    725 sec(s) ago`
`Mode           = AC`
`Power source   = AC`

`+++ Processor`
`CPU model      = Intel(R) Xeon(R) CPU E3-1505M v5 @ 2.80GHz`

AliTanweer, 2016-09-06:
Did you try to configure it ?

http://linrunner.de/en/tlp/docs/tlp-configuration.html

greengene, 2016-09-06:
I have made a few changes in the configuration, but reverted them once I noticed this problem. But the problem has persisted regardless.
If you are trying to say that certain changes are required to make the fan work properly - well, I don't see any such parameter there (BTW that's kind of strange - I would expect an application such as TLP to let me configure the fan behavior).
Please explain what you mean.

AliTanweer, 2016-09-07:
I thought you didn't, try changing the power saving options in the BIOS if you didn't, it helped me on my thinkpad.

greengene, 2016-09-07:
In the BIOS? I think it can be done in the OS level, if I remember right from many other Linux distros. Which power saving options gave this problem and which did not?
P.S. regardless, it should not work like that (bug?)

AliTanweer, 2016-09-07:
Yes, it might be a bug.

linrunner, 2016-09-07:
Hi,

i **always** need the the **full output of tlp-stat** via [Gist](http://gist.github.com/
). 
No support without that.

1. Does this happen on AC or BAT or both? Please provide the corresponding tlp-stat output.
2. Shot in the dark: try to [blacklist nvidia](http://linrunner.de/en/tlp/docs/tlp-faq.html#nvidia)

greengene, 2016-09-10:
Thanks,

Here is the full output :
https://gist.github.com/greengene/e274163eb47e45dc0b7070012c831a7e

Following your suggestion to blacklist Nvidia, I tried to disable eGPU like this:
`RUNTIME_PM_BLACKLIST="01:00.0" `
but it did not solve the problem.

Here is the output of:
`$ lspci -v | perl -ne '/VGA/../^$/ and /VGA|Kern/ and print'`
`01:00.0 VGA compatible controller: NVIDIA Corporation GM107GLM [Quadro M2000M] (rev a2) (prog-if 00 [VGA controller])`
`	Kernel driver in use: nouveau`
`	Kernel modules: nvidiafb, nouveau`

As for AC and BAT - I have noticed that the fan may start running at high speed and never stop when the laptop is working on AC.
But when the laptop is working on BAT the problem does not appear.

Actually, when the laptop is on AC and the fan starts to run at high speed and never stop, I can disconnect the laptop from the power outlet, and within a few seconds the fan goes down to normal speed. Then I can reconnect it to AC, and the fan remains at normal speed, until sometime later when I run some other app that requires high CPU, and then the fan again would go to high speed without stopping (unless I again disconnect it from AC).

linrunner, 2016-09-12:
Reading your description i'm not yet convinced that TLP causes this behaviour. Did you re-test long enough, maybe uninstall TLP?

Btw: i had to use the trick with disconnecting the AC briefly on my X200 years ago.

Things i'd try:

1. Update BIOS to ver 1.30
2. [Mainline kernel](http://kernel.ubuntu.com/~kernel-ppa/mainline/) 4.7 or 4.8rc
3. Disable RUNTIME_PM_ON* completely by commenting both lines

greengene, 2016-09-13:
Do you mean uninstall & re-install TLP?

linrunner, 2016-09-13:
Uninstall – and check if the symptoms occur without TLP.

greengene, 2016-09-13:
I'm going to try some of your other suggestions.

But isn't the fact that the problem is gone when TLP is installed but disabled, a good indication that it's the fact that TLP is enabled that triggers this problem (however it may not be the culprit)?

Also, between upgrading to Ubuntu 16.04.1 and installing TLP, I had about a month in which there was no such problem. It clearly started right after I installed TLP (and had it enabled). So I think this answers your uninstalling suggestion?

linrunner, 2016-09-14:
OK. Go ahead.

closed 2016-11-12 by linrunner

i2000s, 2016-12-07:
```
$lspci -v | perl -ne '/VGA/../^$/ and /VGA|Kern/ and print'
00:02.0 VGA compatible controller: Intel Corporation Skylake Integrated Graphics (rev 06) (prog-if 00 [VGA controller])
	Kernel driver in use: i915
	Kernel modules: i915
01:00.0 VGA compatible controller: NVIDIA Corporation GM107GLM [Quadro M1000M] (rev a2) (prog-if 00 [VGA controller])
	Kernel driver in use: nvidia
	Kernel modules: nvidiafb, nouveau, nvidia_367, nvidia_367_drm
```
Same issue here. Anyway to solve this problem without uninstalling TLP?

peos3, 2018-06-19:
I also ran into this issue. I installed TLP 1.1 with acpi-call-dkms.

**Systeminfo:**
Hardware: ThinkPad P50
BIOS: N1EET74W (1.47 )
System: Ubuntu 18.04
Kernel: 4.15.0

**issue:**
High fan speed (about 3778 rpm) while system is not under load (idle)
Tested in AC mode.

When I uninstall TLP the fan behaves normal.

One thing i noticed:
When checking the temperature sensors, i found the following:

Without TLP installed with fan running at 0 rpm:
```
pch_skylake-virtual-0
Adapter: Virtual device
temp1:        +59.0°C
```

With TLP installed and fan running at nearly 4000 rpm:
```
pch_skylake-virtual-0
Adapter: Virtual device
temp1:        +72.0°C
```

I can't explain the temperature difference though. But it might be the reason for the high fan speed.

greengene, 2018-07-22:
Yes. This problem is still here. It's been for years, with several Thinkpad models that I have tested. I just tested on P50 with BIOS N1EET78W (1.51), and when TLP is installed on enabled, the fan goes above 3770 RPM and never goes down.

propcoder, 2021-03-23:
Exactly the same symptoms like @greengene described on Lenovo ThinkPad P50, BIOS N1EET90W (1.63 ).
[Full tlp-stat output](https://gist.github.com/propcoder/403e737a518134136d17efa45217e161)

domy86, 2022-02-04:
Just tested now on both my and friend of mine T480 with Fedora 35, I thought I'm crazy. After installing TLP, the fan is spinning very long time, even when all temp sensors read 35-40°C. If I disable TLP (just by setting TLP_ENABLE=0) and restart the notebook afterwards, it works fine. During light office work, the fun spins for approx 15-20 seconds without TLP. Actually, this problem bothers me more than one year but I was too lazy to test. Any solution?

domy86, 2022-02-05:
Added RUNTIME_PM_ON_AC="auto" to my custom config and looks OK so far. Googled later on and found a lot of problems related to RUNTIME_PM_ON_AC. Maybe you can put some disclaimer that it can cause some problems with fans, performance etc.

[image]

tonybenoy, 2023-01-15:
I just went through the same problem. The funny thing is fans stop when they are left idle and ramp up to full speed the second I do anything. It stops the second my keyboard backlight turns off after 5 seconds timeout.  I was blaming my Nvidia GPU for all the obvious reasons and I was living the +4 hours of battery life to even question TLP. I finally got frustrated with the fan being always on that I started installing and uninstalling probable offenders. And behold my device is silent after uninstalling TLP. 
I'm not entirely blaming TLP as it could be a config issue(I really didn't tweak anything). I will hopefully reinstall over the next weekend again to see if I can pinpoint something.

linrunner, 2023-01-16:
@tonybenoy try to configure `RUNTIME_PM_ON_AC=auto`. I'm considering to change the default settings accordingly and need more evidence. Also show the full output of **tlp-stat** via https://gist.github.com when done. Thanks.

tonybenoy, 2023-01-22:
@linrunner  I tried with `RUNTIME_PM_ON_AC=auto` . I still have the issue. I did try running `powertop auto-tune` this did seem to happen once with it too but only once and seemed more in line with high cpu temps. Here's the [gist](https://gist.github.com/tonybenoy/8bc4aa3ae790a3f20558124870f149bd). I will experiment further though. Also another thing that I noticed was lower times with `RUNTIME_PM_ON_AC=auto`

tonybenoy, 2023-01-22:
After playing around a bit more it seems the issue was thermald and not tlp or powertop. I've removed thermald and it seems my fans are at a spectrum rather than going full throttle.

linrunner, 2023-01-23:
@tonybenoy thanks for your feedback, added to the FAQ: https://linrunner.de/tlp/faq/powercon.html#high-fan-speed