linrunner/TLP #227: ThinkPad 13 battery threshold problem

Archer785, 2016-09-08:
Hello, I have installed TLP and acpi-call-dkms on my ThinkPad 13, but when I run tlp-stat, it says : 
"tpacpi-bat = inactive (unsupported hardware)"

Are you planning to support it as well or is the problem somewhere else ?
Thank you very much

linrunner, 2016-09-09:
Hi,

i **always** need the the **full output of tlp-stat** via [Gist](http://gist.github.com/). 
No [support](http://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html#support) without that.

Archer785, 2016-09-09:
Hello,
pardon me, but I thought that is what I gave to you in original post. However this is verbose output of tlp-stat :
https://gist.github.com/Archer785/2cd745a31e9d69ede8e6358679fa1396
Thanks

linrunner, 2016-09-10:
Your model "S2" is not in the detection regex. To add it change the file **/usr/share/tlp-pm/tlp-functions**, line 45 to

    readonly RE_TPACPI_ONLY='^((Edge )?E[1345][23456][05][s]?|Helix.*|L[45][3456]0|P[75]0|(Edge )?S[245][345][01]*|S2|T[45][3456][01][psu]?|W5[345][01][s]?|X1 Carbon.*|X2[3456]0[s]?( Tablet)?|.*Yoga.*)$'

The line is quite long, make sure you get the whole string by cut&paste. 

Show the output of

    sudo tlp-stat -b

after the change.

Archer785, 2016-09-10:
Hello, thanks for your help, but I dont think it is enough. tlp-stat -b output : 
/usr/bin/tlp-stat: 1: /etc/default/tlp: k#: not found
--- TLP 0.9 --------------------------------------------

+++ ThinkPad Extended Battery Functions
tp-smapi   = inactive (kernel module 'tp_smapi' not installed)
tpacpi-bat = inactive (unsupported hardware)

+++ Battery Status
/sys/class/power_supply/BAT1/manufacturer                   = SMP
/sys/class/power_supply/BAT1/model_name                     = LNV-01AV401
/sys/class/power_supply/BAT1/cycle_count                    = (not supported)
/sys/class/power_supply/BAT1/energy_full_design             =  42010 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  45840 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  39430 [mWh]
/sys/class/power_supply/BAT1/power_now                      =   5364 [mW]
/sys/class/power_supply/BAT1/status                         = Discharging

Charge                                                      =   86.0 [%]
Capacity                                                    =  109.1 [%]

linrunner, 2016-09-10:
May be your edit has gone wrong. I made a package for you. 

Download [tlp_0.9.001-1_all.deb](http://download.linrunner.de/packages/tlp_0.9.001-1_all.deb) and install with dpkg -i

Archer785, 2016-09-10:
thanks, but the output is still the same (except version number)
/usr/bin/tlp-stat: 1: /etc/default/tlp: k#: not found
--- TLP 0.9.001 --------------------------------------------

+++ ThinkPad Extended Battery Functions
tp-smapi   = inactive (kernel module 'tp_smapi' not installed)
tpacpi-bat = inactive (unsupported hardware)

+++ Battery Status
/sys/class/power_supply/BAT1/manufacturer                   = SMP
/sys/class/power_supply/BAT1/model_name                     = LNV-01AV401
/sys/class/power_supply/BAT1/cycle_count                    = (not supported)
/sys/class/power_supply/BAT1/energy_full_design             =  42010 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  45840 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  38690 [mWh]
/sys/class/power_supply/BAT1/power_now                      =   9762 [mW]
/sys/class/power_supply/BAT1/status                         = Discharging

linrunner, 2016-09-10:
Show the output of

    sudo /usr/share/tlp-pm/tpacpi-bat -g ST 1

Archer785, 2016-09-10:
Error: AE_NOT_FOUND for ASL base: \_SB.HKEY
Error: AE_NOT_FOUND at /usr/share/tlp-pm/tpacpi-bat line 409.

linrunner, 2016-09-10:
Then you're out of luck because Lenovo removed the necessary functions from ACPI. Sorry.

closed 2016-09-10 by linrunner

theonedemon, 2017-01-23:
I have some problems:

--- TLP 0.9 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
TLP_DEFAULT_MODE=BAT
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
CPU_SCALING_GOVERNOR_ON_AC=performance
CPU_SCALING_GOVERNOR_ON_BAT=powersave
CPU_MIN_PERF_ON_AC=0
CPU_MAX_PERF_ON_AC=100
CPU_MIN_PERF_ON_BAT=0
CPU_MAX_PERF_ON_BAT=50
CPU_BOOST_ON_AC=1
CPU_BOOST_ON_BAT=0
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau"
USB_AUTOSUSPEND=1
USB_BLACKLIST="1d57:5a66"
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0
DEVICES_TO_DISABLE_ON_STARTUP="bluetooth wwan"
DEVICES_TO_ENABLE_ON_STARTUP="wifi"

+++ System Info
System         = LENOVO Lenovo IdeaPad Y500 20193
BIOS           = 6BCN44WW(V2.04)
Release        = Ubuntu 16.04.1 LTS
Kernel         = 4.4.0-57-generic #78-Ubuntu SMP Fri Dec 9 23:50:32 UTC 2016 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-4.4.0-57-generic.efi.signed root=UUID=cc0ec811-5aa3-489c-bad6-7f79b4a56c69 ro quiet nosplash intel_pstate=enable
Init system    = systemd v229
Boot mode      = UEFI

+++ TLP Status
State          = enabled
Last run       = 11:07:39,    579 sec(s) ago
Mode           = AC
Power source   = AC

+++ Processor
CPU model      = Intel(R) Core(TM) i5-3230M CPU @ 2.60GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = performance
[... 154 lines cut ...]
Bus 004 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 003 Device 002 ID 174f:1474 control = auto, autosuspend_delay_ms =  2000 -- Syntek  (uvcvideo)
Bus 003 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)

+++ Battery Status
/sys/class/power_supply/BAT1/manufacturer                   = LENOVO 
/sys/class/power_supply/BAT1/model_name                     = PABAS0241231
/sys/class/power_supply/BAT1/cycle_count                    = (not supported)
/sys/class/power_supply/BAT1/energy_full_design             =  62710 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  61550 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  61550 [mWh]
/sys/class/power_supply/BAT1/power_now                      = 655350 [mW]
/sys/class/power_supply/BAT1/status                         = Full

Charge                                                      =  100.0 [%]
Capacity                                                    =   98.2 [%]

In dmesg message:
[ 7952.158248] acpi_call: Cannot get handle: Error: AE_NOT_FOUND
[ 7952.854462] acpi_call: Cannot get handle: Error: AE_NOT_FOUND
[ 7952.882246] acpi_call: Cannot get handle: Error: AE_NOT_FOUND

journalctl:
kernel: acpi_call: Cannot get handle: Error: AE_NOT_FOUND
kernel: acpi_call: Cannot get handle: Error: AE_NOT_FOUND

linrunner, 2017-01-23:
What problem? ThinkPad battery features are not supported on Ideapads. I think he [FAQ](http://linrunner.de/en/tlp/docs/tlp-faq.html#battery) is quite clear on that.

As soon as you uninstall the package **acpi-call-dkms** those messages will disappear

RX14, 2017-11-27:
Hi, this is no longer the case: I've just submitted a patch to upstream `tpacpi-bat` which adds support for thinkpad 13: https://github.com/teleshoes/tpacpi-bat/pull/90.

Archer785, 2017-11-27:
@RX14 that is great ! when can we expect it to be available via apt ?

RX14, 2017-11-27:
Depends on a lot of things: my patch getting merged, TLP updating it's blacklist, a new release of tpacpi-bat and TLP, and then whoever does the packaging (i don't use debian/ubuntu i'm not sure if this is a distro package or custom repo) has to update that and then it has to get into the repos.