linrunner/TLP #228: VM writeback timeout and Touch Controllers aren't properly tuned

3v1n0, 2016-09-10:
I'm using TLP for some time and I like it a lot, however when TLP is running and I open powertop, there are two tunings that aren't properly set in my hardware (T460s with touch screen):
```
>> Bad      VM writeback timeout                                                                               
>> Bad      Autosuspend for USB device LGD AIT Touch Controller [Melfas]
```
Powertop adjusts these values with these commands:
```bash
echo '1500' > '/proc/sys/vm/dirty_writeback_centisecs'
echo 'auto' > '/sys/bus/usb/devices/1-10/power/control'
```

So I was wondering why can't be done by TLP, or if there are some settings preventing this to happen.

linrunner, 2016-09-10:
Hi, please read the [FAQ](http://linrunner.de/en/tlp/docs/tlp-faq.html#powertop) before opening issues. Thank you.

closed 2016-09-10 by linrunner

3v1n0, 2016-09-11:
Sorry I didn't check that bafore... :-(