linrunner/TLP #233: High power consumption when using intel onboard skylake graphics

jimitsalvi, 2016-10-07:
### My battery is reporting high power usage while using Intel onboard graphics. When I switch to my NVIDIA card the power consumption reduces significantly. I am using powertop just for monitoring my power usage. I am a newbie to linux and github so kindly forgive me if this issue is already posted before.

1. It occurs mainly on battery use. Powertop doesn't provide the power consumption while on AC.
2. Output while using onboard intel graphics
--- TLP 0.8 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
TLP_DEFAULT_MODE=AC
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau"
USB_AUTOSUSPEND=1
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0

+++ System Info
System         = ASUSTeK COMPUTER INC. 1.0 G501VW
BIOS           = G501VW.301
Release        = Linux Mint 18 Sarah
Kernel         = 4.8.1-040801-generic #201610071031 SMP Fri Oct 7 14:34:10 UTC 2016 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-4.8.1-040801-generic root=UUID=80768e65-ce86-4749-bc24-a88e9f990875 ro acpi_osi=! acpi_backlight=native quiet splash vt.handoff=7
Init system    = systemd

+++ System Status
TLP power save = enabled
power source   = battery

+++ Processor
CPU Model      = Intel(R) Core(TM) i7-6700HQ CPU @ 2.60GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   800000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  3500000 [kHz]

/sys/devices/system/cpu/cpu1/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu1/cpufreq/scaling_min_freq  =   800000 [kHz]
[... 325 lines cut ...]
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)

+++ Battery Status
/sys/class/power_supply/BAT0/manufacturer                   = ASUSTeK
/sys/class/power_supply/BAT0/model_name                     = ASUS Battery
/sys/class/power_supply/BAT0/cycle_count                    =     75
/sys/class/power_supply/BAT0/energy_full_design             =  60040 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  53063 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  10275 [mWh]
/sys/class/power_supply/BAT0/power_now                      =  11673 [mW]
/sys/class/power_supply/BAT0/status                         = Discharging

+++ Suggestions
* Install smartmontools for dis

### Power Consumption while using intel onboard graphic card should be less or atleast at par with NVIDIA card usage.

### Battery reports high power usage as much as 4watts while using onboard graphic card.

### Use intel onboard graphic card and measure active power usage using powerstat(or any other tool). Switch to NVIDIA graphic card and measure the active power usage.

I hope this helps.

[image]
[image]

linrunner, 2016-10-07:
Hi, 

in fact TLP does *not* contain any power saving features for Intel graphics. The kernel driver i915 itself enables power saving upon boot. So i doubt your problem is related to TLP at all.

What happens when you uninstall TLP and reboot – does the power consumption decrease?

jimitsalvi, 2016-10-07:
Hello,
Thank you for the prompt response. 
Yes It Does. I uninstalled TLP and this is what I found. I could see some reduction in power consumption compared to TLP installed (running on Intel Onboard Graphics)

[image]

linrunner, 2016-10-07:
Please stop posting screenshots. The power consumption is shown in the output of tlp-stat:

     tlp-stat -b | grep power_now

**You have to wait 2-3 minutes with idle desktop – all applications closed – to get comparable readings!**

Try to change the following configuration line as shown:

    RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau i915"
    
Reboot.

jimitsalvi, 2016-10-07:
My apologies.
I followed your instructions.
/sys/class/power_supply/BAT0/power_now                      =  13467 [mW]

linrunner, 2016-10-07:
Apart from the above config line i have no more suggestions for you.

Hybrid graphics isn't fun with Linux. Sorry.

ArchangeGabriel, 2016-10-08:
@jimitsalvi Do you use nouveau or bbswitch for dedicated card power management?

jimitsalvi, 2016-10-08:
@linrunner - No issues. I appreciate your help. I hope linux developers would fix this issue down the road.
@ArchangeGabriel - No, I just use the regular NVIDIA driver (NVIDIA X Server) from the driver manager. No power management tweaks applied apart from the TLP installation. I had blacklisted nouveau driver during installation because of boot issues..
I did uninstall the xserver-xorg-video-intel driver because it discouraged its use on newer hardware in its notes(however uninstallation did not make difference as the power consumption remained same).

khfeng, 2016-10-08:
I think the issue here is that the driver was still loaded, so bbswitch cannot turn it off:
/sys/bus/pci/devices/0000:01:00.0/power/control = auto (0x030200, 3D controller, nvidia)
This should be no driver instead of nvidia.

This happens because newer Nvidia driver has an additional kernel module that needs to unload. This should be fixed at [1].  

If that alone doesn't work, try appending acpi_osi="Windows 2013" after acpi_osi=!.

[1] https://github.com/tseliot/ubuntu-drivers-common/commit/b80261ecaf2584eddc0bc5a3ab12e3c054bea49c

ArchangeGabriel, 2016-10-08:
@khfeng I’ve seen nowhere @jimitsalvi stating he was using bbswitch/Bumblebee. Quite the contrary in fact. ;)

@jimitsalvi That’s your issue. You have to use one module for Nvidia card power management, else what you see is expected. Currently, your situation is the following:

1. When running on the integrated card, both cards are in fact consuming power, with the Nvidia one consuming at its idle rate (actually, the Intel one too).
2. When running on the dedicated card, the Intel one is almost shut down, and the Nvidia one is consuming at its idle rate again.

To solve this, when running on the Intel card, you must either use bbswitch (eventually through Bumblebee) or nouveau to handle power management of the dedicated card.

khfeng, 2016-10-08:
@ArchangeGabriel it's pure guessing... gpu-manager (which uses bbswitch) should be used if Linux Mint doesn't deviate from Ubuntu too much.

ArchangeGabriel, 2016-10-08:
@khfeng I think that Mint uses Ubuntu repos… and that’s all. ;)

jimitsalvi, 2016-10-09:
@khfeng - I wasn't using bbswitch.
@ArchangeGabriel - I tried installing bumblebee, but when I restarted, I would get an error stating no display drivers found and mdm would not start (failed to start X server).
Error code:
mdm.service: Control process exited, code=exited status=203.
Fatal error: No screens found.

I also tried using nouveau driver but my power consumption remains about 14w vs Nvidia 9w.

jimitsalvi, 2016-10-09:
@ArchangeGabriel - I got the bumblebee to work. This is what it says, if it helps.

dmesg | grep bbswitch
[    4.268278] bbswitch: loading out-of-tree module taints kernel.
[    4.268292] bbswitch: module verification failed: signature and/or required key missing - tainting kernel
[    4.268385] bbswitch: version 0.8
[    4.268388] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[    4.268391] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG0.PEGP
[    4.268451] bbswitch: detected an Optimus _DSM function
[    4.268520] bbswitch: Succesfully loaded. Discrete card 0000:01:00.0 is on
[    4.698432]  idma64 snd mei_me virt_dma processor_thermal_device soundcore mei intel_lpss_pci shpchp intel_soc_dts_iosf elan_i2c int3403_thermal bluetooth rfkill wmi battery intel_lpss_acpi int3402_thermal intel_lpss tpm_crb int340x_thermal_zone int3406_thermal ac int3400_thermal acpi_pad acpi_thermal_rel asus_wireless evdev ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 xt_hl ip6t_rt nf_conntrack_ipv6 nf_defrag_ipv6 ipt_REJECT nf_reject_ipv4 nf_log_ipv4 nf_log_common xt_LOG xt_limit xt_tcpudp xt_addrtype nf_conntrack_ipv4 nf_defrag_ipv4 xt_conntrack ip6table_filter ip6_tables nf_conntrack_netbios_ns nf_conntrack_broadcast nf_nat_ftp nf_nat bbswitch(OE) nf_conntrack_ftp parport_pc nf_conntrack ppdev iptable_filter lp ip_tables parport x_tables efivarfs autofs4 ext4 crc16 jbd2 fscrypto mbcache btrfs xor raid6_pq

ArchangeGabriel, 2016-10-09:
@jimitsalvi This excerpt is too short, please gist the full dmesg if possible. Also, what returns `cat /proc/acpi/bbswitch`? FYI, the issue you had with non-starting X was because you let it in the Nvidia configuration prior to Bumblebee installation, which isn’t supported currently.

Depending on your next message, we might migrate to [bbswitch tracker][1].

[1]:https://github.com/Bumblebee-Project/bbswitch/issues

jimitsalvi, 2016-10-09:
@ArchangeGabriel - Here is the output of dmesg.
[    0.000000] Linux version 4.8.1-040801-generic (kernel@tangerine) (gcc version 6.2.0 20160927 (Ubuntu 6.2.0-5ubuntu11) ) #201610071031 SMP Fri Oct 7 14:34:10 UTC 2016
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.8.1-040801-generic root=UUID=80768e65-ce86-4749-bc24-a88e9f990875 ro acpi_osi=! acpi_backlight=native quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] x86/fpu: Using 'eager' FPU context switches.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000327eafff] usable
[    0.000000] BIOS-e820: [mem 0x00000000327eb000-0x00000000327ebfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000327ec000-0x0000000032835fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000032836000-0x00000000354f6fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000354f7000-0x0000000036219fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003621a000-0x00000000364f2fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000364f3000-0x000000003709ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000370a0000-0x0000000037f01fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000037f02000-0x0000000037f7efff] type 20
[    0.000000] BIOS-e820: [mem 0x0000000037f7f000-0x0000000037ffefff] usable
[    0.000000] BIOS-e820: [mem 0x0000000038000000-0x00000000380fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000006c1ffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.40 by American Megatrends
[    0.000000] efi:  ESRT=0x37e41e18  ACPI=0x365fb000  ACPI 2.0=0x365fb000  SMBIOS=0x3765b000  SMBIOS 3.0=0x3765a000 
[    0.000000] esrt: Reserving ESRT space from 0x0000000037e41e18 to 0x0000000037e41e50.
[    0.000000] SMBIOS 3.0.0 present.
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x6c2000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: write-back
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 0080000000 mask 7F80000000 uncachable
[    0.000000]   1 base 0040000000 mask 7FC0000000 uncachable
[    0.000000]   2 base 003C000000 mask 7FFC000000 uncachable
[    0.000000]   3 base 003A000000 mask 7FFE000000 uncachable
[    0.000000]   4 base 0039000000 mask 7FFF000000 uncachable
[    0.000000]   5 base 0038800000 mask 7FFF800000 uncachable
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000]   8 disabled
[    0.000000]   9 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT  
[    0.000000] e820: last_pfn = 0x37fff max_arch_pfn = 0x400000000
[    0.000000] Base memory trampoline at [ffff895040098000] 98000 size 24576
[    0.000000] Using GB pages for direct mapping
[    0.000000] BRK [0x648452000, 0x648452fff] PGTABLE
[    0.000000] BRK [0x648453000, 0x648453fff] PGTABLE
[    0.000000] BRK [0x648454000, 0x648454fff] PGTABLE
[... 1045 lines cut ...]
[  615.719729] Bluetooth: RFCOMM TTY layer initialized
[  615.719737] Bluetooth: RFCOMM socket layer initialized
[  615.719746] Bluetooth: RFCOMM ver 1.11
[  615.841516] [drm] RC6 on
[  628.976003] iwlwifi 0000:03:00.0: L1 Enabled - LTR Enabled
[  628.976544] iwlwifi 0000:03:00.0: L1 Enabled - LTR Enabled
[  629.044684] iwlwifi 0000:03:00.0: L1 Enabled - LTR Enabled
[  629.045433] iwlwifi 0000:03:00.0: L1 Enabled - LTR Enabled
[  629.073337] IPv6: ADDRCONF(NETDEV_UP): wlp3s0: link is not ready
[  629.150145] IPv6: ADDRCONF(NETDEV_UP): wlp3s0: link is not ready
[  632.291328] wlp3s0: authenticate with 48:5d:36:a6:26:ca
[  632.296686] wlp3s0: send auth to 48:5d:36:a6:26:ca (try 1/3)
[  632.297372] wlp3s0: authenticated
[... cut ...]
[  632.394756] wlp3s0: Limiting TX power to 30 (30 - 0) dBm as advertised by 48:5d:36:a6:26:ca
[  695.269234] [UFW BLOCK] IN=wlp3s0 OUT= MAC=01:00:5e:00:00:01:48:5d:36:a6:26:c9:08:00 SRC=192.168.1.1 DST=224.0.0.1 LEN=36 TOS=0x00 PREC=0xC0 TTL=1 ID=63740 DF PROTO=2 
[  727.014750] [UFW BLOCK] IN=wlp3s0 OUT= MAC=01:00:5e:00:00:fb:a4:b8:05:aa:ce:2f:08:00 SRC=192.168.1.175 DST=224.0.0.251 LEN=32 TOS=0x00 PREC=0x00 TTL=1 ID=45374 PROTO=2 

cat /proc/acpi/bbswitch 
0000:01:00.0 ON

The bbswitch output shows that the NVIDIA card is running.

ArchangeGabriel, 2016-10-09:
Please don’t past dmesg output directly in the GitHub issue, it’s really ugly and impractical. Use gist or a paste service for that.

So, you’re not using Bumblebee currently, are you? The Nvidia card is indeed still powered on because of the nvidia driver using it. You need to blacklist nvidia, nvidia-uvm, nvidia-drm and nvidia-modeset (through a file in `/etc/modprobe.d`, don’t forget to update initramfs) or manually remove them one by one (using `modprobe -r <modulename>`, use `lsmod | grep nvidia` in order to get the right order — remove those used by no other first).

jimitsalvi, 2016-10-09:
@ArchangeGabriel - My apologies.
https://gist.github.com/jimitsalvi/fcec572d8e0594ccae154b1e8d6d212c

I added the following lines in /etc/modprobe.d/Bumblebee.conf
# 367
blacklist nvidia-367
blacklist nvidia-367-updates
blacklist nvidia-experimental-367
# Workaround to make sure nvidia-uvm is removed as well
remove nvidia rmmod nvidia-drm nvidia-modeset nvidia-uvm nvidia

This is what I got after updating initramfs
lsmod | grep nvidia
nvidia_drm             45056  0
nvidia_modeset        765952  1 nvidia_drm
nvidia              11472896  1 nvidia_modeset
drm_kms_helper        167936  2 i915,nvidia_drm
drm                   368640  7 i915,nvidia_drm,drm_kms_helper

PS: Please bear with me, I have just switched from windows 10 to linux and github and trying to get a hold of it. I appreciate your help.

ArchangeGabriel, 2016-10-09:
OK. Let’s keep the blacklist thing aside for now. What if you try this:
```
sudo modprobe -r nvidia_drm
sudo modprobe -r nvidia_modeset
sudo modprobe -r nvidia
sudo tee /proc/acpi/bbswitch <<< OFF
```

Do you see any benefit in power consumption?

jimitsalvi, 2016-10-09:
@ArchangeGabriel - Yes I do see the benifit. Power consumption dropped to 11.5 vs 14.5.

ArchangeGabriel, 2016-10-09:
OK, nice. For better results, you need PCIe port PM, which isn’t available yet, and better power saving options for i915 (psr, fbc, rc6pp, dc…).

Now, your possibilities depends on your usage. Theoretically, everything is possible, but a lot of setups require a bunch of configs. What you can easily do is one of the following:

1. Use the Intel card for display, run 3D-heavy programs on the Nvidia one using nouveau through DRI_PRIME. This probably gives the best power consumption, but the performances are not good and won’t be until nouveau support proper reclocking.

2. Use the Intel card for display, run 3D-heavy programs on the Nvidia one using optirun (part of Bumblebee, which also handles power management through bbswitch). Perfs are a bit better than 1., but still not ideal.

3. Use only the Nvidia card. If you’re really getting the lowest power consumption with this setup (which is strange BTW), this is probably your best go. I would recommend trying Intel with all i915 options first.

For i915 options, add a file under `/etc/modprobe.d/` with this:
`options i915 enable_rc6=7 enable_fbc=1 enable_dc=2 semaphores=1 modeset=1 enable_psr=1 fastboot=Y`

Rebuild your initramfs (`sudo update-initramfs -u`) and reboot. You might observe flickering and other unwanted effects (which are among the reason those options aren’t default ones).

jimitsalvi, 2016-10-09:
@ArchangeGabriel - I tried the third option. It broke my system with random freezes as you told. I had to uninstall bumblebee and nvidia to get back into my system.

ArchangeGabriel, 2016-10-17:
Sorry for the delay…

Then I think your best option is your initial setup if you’re really experiencing such a low consumption on nvidia only.

linrunner, 2017-01-04:
Issue abandoned --> CLOSE

closed 2017-01-04 by linrunner