linrunner/TLP #234: wired ethernet connection cannot connect after warm reboot

alvinh16, 2016-10-25:
### Symptom data
Upon installing tlp & rebooting, wired ethernet cannot connect. 
removing tlp does not help. The system still does not connect.

1. Does the problem occur on battery or AC or both? both
2. Attach the *full* output of `tlp-stat` via [Gist](https://gist.github.com/) for *all* cases of 1. Can't, it's tlp that causes the network not to work. 

### Expected behavior
upon installing tlp & doing a (warm) reboot, eg after update, the wired network should connect as before installation of tlp. 

### Actual behavior
wired network cannot connect. 
curious observation : removing tlp does not work. It still cannot connect. 

my thots : there are 2 problems here;
1. the installation procedure of tlp disables the wired ethernet connection, not necessarily tlp itself. The hint came from here;
http://askubuntu.com/questions/660164/lan-not-working-after-reboot

2. upon removing tlp, it should return the system to how it was before ie wired ethernet should still be able to connect. But it can't!!! 

conclusion, 
1. the installation process changes something that causes wired ethernet not to be able to connect. 
2. the uninstallation/deinstallation process does not return the system to the original state before the installation 

if, however, i absolutely do not install tllp, the system works beautifully. Between reboots, i can connect to ethernet no matter how many times i reboot, regardless also of whatever kind of update or software installation. 

### Steps to reproduce the problem
OS : Ubuntu 16.10

This porblem only manifests itself on certain hardware. I have 2 other laptops which do not have this problem, but they are generation 2 CPUs. This is a gen 6 (skylake). Here are my hardware specs
Manufacturer : Acer
Product name : Aspire E14
model : E5-474G-53G4

$ sudo lshw -C network
[sudo] password for alvinh: 
  *-network                 
       description: Ethernet interface
       product: RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0.1
       bus info: pci@0000:02:00.1
       logical name: enp2s0f1
       version: 12
       serial: f0:76:1c:ce:2d:1a
       size: 1Gbit/s
       capacity: 1Gbit/s
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress msix vpd bus_master cap_list ethernet physical tp mii 10bt 10bt-fd 100bt 100bt-fd 1000bt 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=r8169 driverversion=2.3LK-NAPI duplex=full firmware=rtl8411-2_0.0.1 07/08/13 ip=192.168.1.65 latency=0 link=yes multicast=yes port=MII speed=1Gbit/s
       resources: irq:276 ioport:3000(size=256) memory:94404000-94404fff memory:94400000-94403fff
  *-network
       description: Wireless interface
       product: QCA9377 802.11ac Wireless Network Adapter
       vendor: Qualcomm Atheros
       physical id: 0
       bus info: pci@0000:03:00.0
       logical name: wlp3s0
       version: 30
       serial: 30:52:cb:2c:cd:13
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress bus_master cap_list ethernet physical wireless
       configuration: broadcast=yes driver=ath10k_pci driverversion=4.8.0-26-generic firmware=WLAN.TF.1.0-00267-1 ip=192.168.1.69 latency=0 link=yes multicast=yes wireless=IEEE 802.11
       resources: irq:281 memory:94200000-943fffff

$ ifconfig
enp2s0f1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.65  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::9c12:2e17:fae5:d459  prefixlen 64  scopeid 0x20<link>
        ether f0:76:1c:ce:2d:1a  txqueuelen 1000  (Ethernet)
        RX packets 13264  bytes 13391857 (13.3 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 6884  bytes 866812 (866.8 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 22489  bytes 1405097 (1.4 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 22489  bytes 1405097 (1.4 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlp3s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.69  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::bf68:3183:13b7:f466  prefixlen 64  scopeid 0x20<link>
        ether 30:52:cb:2c:cd:13  txqueuelen 1000  (Ethernet)
        RX packets 1210  bytes 137681 (137.6 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 448  bytes 48224 (48.2 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

linrunner, 2016-10-25:
> Attach the full output of tlp-stat via Gist for all cases of. Can't, it's tlp that causes the network not to work. 

Yes, you can. Hint: redirect the output to a file. 

The installation process doesn't apply any settings, just copies files and enables the services but doesn't start them immediately. Uninstall just removes the files and services. You can remove the remaining config files with

    sudo apt-get purge tlp tlp-rdw

You may check the installed files at any time with

    dpkg -L tlp tlp-rdw

If you uninstalled with _remove_ instead of _purge_ you will see the following files

    /etc/default/tlp
    /etc/init.d/tlp 
    /etc/NetworkManager/dispatcher.d/99tlp-rdw-nm 

These files alone are not functional and do no harm to your system.

alvinh16, 2016-10-25:
dear linerunner,

u don't understand. i've already reinstalled ubuntu 16.10 more than 5 
times in order to narrow down that tlp was the problem. Do u really need 
this info? if so, watt do u need it for? i'm willing 2 install another 
partition for you if you can substantiate that this is something that 
you really need & not just a formality.

alf
Alvin Ho dtm
alvinho@yandex.com
+65-9688-3129

Secretary
Nee Soon East Toastmasters Club
http://neesooneast.toastmastersclubs.org/

Put first things first and we get second things thrown in:
put second things first and we lose both first and second things. - C S 
Lewis

Sent from my Ubuntu Linux 16.04 desktop

linrunner, 2016-10-25:
How did you expect to isolate the problem without having an installation actually showing the symptom? That's impossible. So a parallel install in a separate partition is a good idea. There is always considerable mileage on behalf of the user when doing remote analysis. Arguing about the support strategy doesn't help. 

It's your decision if you want to solve your problem – and maybe improve TLP.

alvinh16, 2016-10-26:
okay i hope this was worth it.. i have installed a partition SPECIALLY for THIS tlp problem. Honestly i don't understand the output. i'll c if the procedure to undo the tlp installation works. i don't really have a lot of faith in it, but will let u know if it works/doesn't anyway.

alvinh16, 2016-10-26:
hi linerunner, i ran the commandsudo apt-get purge tlp tlp-rdw it's still in the command history, so i did not make a mistake. moreover, i did a copy & paste. As u mentioned, the 3 files;/etc/default/tlp /etc/init.d/tlp /etc/NetworkManager/dispatcher.d/99tlp-rdw-nm  are gone. As I mentioned.. the system STILL cannot connect after a warm boot. I hate to say "i told you so.." but i told you so. it's not tlp's "fault" per se, but the installation procedure & the removal procedure does not restore the system to its original state. So i'm stuck with a system that cannot connect to wired network after a warm reboot iesudo shutdown -r 0 okay, i did everything that you asked of me. The balls in your court now i'm afraid.

linrunner, 2016-10-26:
I'm still missing the full tlp-stat output.

Please stop spamming this issue with your mail signature and full quotes. This is caused by answering by mail instead of using the website.

alvinh16, 2016-10-26:
Dono if this works, i have a text file that i used "script" command to produce;
This is while on AC power
Script started on Wed 26 Oct 2016 11:44:41 PM
]0;alvinh@aspire474g: ~/Documents[01;32malvinh@aspire474g[00m:[01;34m~/Documents[00m$ sudo tlp-stat
[sudo] password for alvinh: 
--- TLP 0.9 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
TLP_DEFAULT_MODE=AC
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
AHCI_RUNTIME_PM_TIMEOUT=15
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=off
WIFI_PWR_ON_BAT=on
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau"
USB_AUTOSUSPEND=1
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0

+++ System Info
System         = Acer V1.01 Aspire E5-474G
BIOS           = V1.01
Release        = Ubuntu 16.10
Kernel         = 4.8.0-26-generic #28-Ubuntu SMP Tue Oct 18 14:39:52 UTC 2016 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-4.8.0-26-generic root=UUID=22987d61-3890-481c-97f9-91099ca83575 ro quiet splash vt.handoff=7
Init system    = systemd v231
Boot mode      = BIOS (CSM, Legacy)

+++ TLP Status
State          = enabled
Last run       = 23:33:30,    683 sec(s) ago
Mode           = AC
Power source   = AC

+++ Processor
CPU model      = Intel(R) Core(TM) i5-6200U CPU @ 2.30GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors = performance powersave
[... 365 lines cut ...]
Bus 002 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 001 Device 004 ID 04f2:b520 control = auto, autosuspend_delay_ms =  2000 -- Chicony Electronics Co., Ltd  (uvcvideo)
Bus 001 Device 003 ID 04ca:3015 control = auto, autosuspend_delay_ms =  2000 -- Lite-On Technology Corp.  (btusb)
Bus 001 Device 002 ID 093a:2510 control = on,   autosuspend_delay_ms =  2000 -- Pixart Imaging, Inc. Optical Mouse (usbhid)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)

+++ Battery Status
/sys/class/power_supply/BAT1/manufacturer                   = COMPAL 
/sys/class/power_supply/BAT1/model_name                     = PABAS0241231
/sys/class/power_supply/BAT1/cycle_count                    = (not supported)
/sys/class/power_supply/BAT1/charge_full_design             =   2500 [mAh]
/sys/class/power_supply/BAT1/charge_full                    =   2471 [mAh]
/sys/class/power_supply/BAT1/charge_now                     =   2457 [mAh]
/sys/class/power_supply/BAT1/current_now                    =    595 [mA]
/sys/class/power_supply/BAT1/status                         = Discharging

Charge                                                      =   99.4 [%]
Capacity                                                    =   98.8 [%]

]0;alvinh@aspire474g: ~/Documents[01;32malvinh@aspire474g[00m:[01;34m~/Documents[00m$ exit
exit

Script done on Wed 26 Oct 2016 11:45:20 PM

hope that helped

linrunner, 2016-10-27:
I suggest to blacklist your LAN adapter, e.g.

> /sys/bus/pci/devices/0000:02:00.1/power/control = auto (0x020000, Ethernet controller, r8169)

and enable Wake-on-LAN in the [configuration](http://linrunner.de/en/tlp/docs/tlp-configuration.html):

    RUNTIME_PM_BLACKLIST="02:00.1"

    WOL_DISABLE=Y

Don't miss to remove the leading '#' from the first line above.

Poweroff and restart.

linrunner, 2016-10-28:
Second line changed to WOL_DISABLE=N

alvinh16, 2016-10-28:
sorry, u lost me. Which file are you describing? 
if you're talking about /etc/default/tlp
forget it! You asked me to purge it remember? You asked me to do this
sudo apt-get purge tlp tlp-rdw

which will purge off these files
/etc/default/tlp
/etc/init.d/tlp 
/etc/NetworkManager/dispatcher.d/99tlp-rdw-nm 

As i mentioned earlier, after purging these files, the system should be restored to it's state before the installation of tlp. Apparently, that's not happening. Before the installation, i was able to do a warm boot & the wired LAN came up normally

During the installation, something changed in the configuration of the system that caused the wired LAN not to connect. 

following the purge of those 3 files, a warm warm boot ie
sudo shutdown -r 0

wired LAN is still unable to connect. In short, whatever configuration changes happened during the installation were not restored at removal nor purge of tlp.

linrunner, 2016-10-28:
Fine. I forget it.

closed 2016-10-28 by linrunner