linrunner/TLP #237: WiFi powersave not being set at boot

rafwes, 2016-11-17:
### Symptom data

[tlp-stat](https://gist.github.com/rafwes/6fde28cd437be033d283a9ea0af7f140)
[lspci -nnv](https://gist.github.com/rafwes/a782505e4d58b683d4ef31987b6bc91e)

> `cat /etc/modprobe.d/ath9k.conf ` 
`options ath9k ps_enable=1`

>`iw dev wlp3s0 get power_save `
 `Power save: off `

### Expected behavior

WiFi powersave should be on after boot on battery power.

### Actual behavior

After boot on battery power WiFi powersave will remain off. 

If I restart tlp.service via systemd after boot or connect and disconnect the power adapter, powersave will be correctly set.

### Steps to reproduce the problem

Set `ps_enable = 1` on ath9k module.
Boot system on battery.

linrunner, 2016-11-17:
Hi,

it's kernel / driver (ath9k) problem. I think the following happens:

1. TLP enables wifi power save on boot
2. Wifi connects
3. Power save gets disabled by the driver upon connect (bug!)

TLP has no hook to re-enable wifi power save upon connect. When you restart the service – you should use `tlp start` instead! – or switch power source, TLP enables power save again. 

I suggest you file a kernel bug against ath9k.

rafwes, 2016-11-17:
I did some testing to narrow down a kernel bug report but could not see the pattern you describe.

First, I removed all connections from NetworkManager to ensure that no connection would take place on boot. Even though there was no WiFi connected, powersave was still off. 

Then I masked all NetworkManager units to ensure the interface remained "down" after booting, but this did not help either.

linrunner, 2016-11-18:
Please create a trace for the boot process without NM --> [Trace Mode](http://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html#trace)

rafwes, 2016-11-22:
Sorry for the delay. There you go: [tlp-stat -T](https://gist.github.com/rafwes/a8cf92f82c8a439d18a8c1765a327c46) (with NetworkManager masked).

linrunner, 2016-11-23:
[set_wifi_power_mode()](https://github.com/linrunner/TLP/blob/0.9/tlp-functions.in#L1466) seems to exit prematurely during startup.

Please add some debug code in **/usr/share/tlp-pm/tlp-functions** below [Line 1491](https://github.com/linrunner/TLP/blob/0.9/tlp-functions.in#L1491):

    echo_debug "pm" "set_wifi_power_mode($1).get_wifi_ifaces: wifaces=$wifaces"

and show the trace. I need only the startup phase, not the preceding shutdown.

rafwes, 2016-11-24:
I added the said line to the file, masked NM and then ran "tlp-stat -T". [Here the output](https://gist.github.com/rafwes/9bc8728dadf1aa3b8e55e8061c32b05a)

linrunner, 2016-11-24:
Seems the interface simply isn't there at runtime of set_wifi_power_mode().

Let's add some more debug output in [get_wifi_ifaces()](https://github.com/linrunner/TLP/blob/0.9/tlp-functions.in#L1438) below [Line #1440](https://github.com/linrunner/TLP/blob/0.9/tlp-functions.in#L1440):

    echo_debug "pm" "get_wifi_ifaces.modules: $(ls /sys/module | grep ath)"
    echo_debug "pm" "get_wifi_ifaces.interfaces: $(grep '.*' /sys/class/net/*/uevent)"

Trace again.

rafwes, 2016-11-26:
[New Trace as asked](https://gist.github.com/rafwes/c04eed3c32d4cf3e99e77e9cf9c381b0)

linrunner, 2016-11-27:
The ath9k module is loaded, but the interface not (yet) there, so TLP can't apply powersave. I have no clue why the interface appears so late in the boot process. 

Did you check for trouble with

    dmesg | grep ath9

Do you have some special network configuration (virtual wifi interfaces)?

rafwes, 2016-11-28:
I use ArchLinux and ath9k gets autodetected as shown with the command `mkinitcpio -M.` That's all I can say. Could you please give me any pointers on how to check if it is really in initramfs? 

Output from dmesg:

```
[    5.482245] ath: phy0: ASPM enabled: 0x42
[    5.482247] ath: EEPROM regdomain: 0x60
[    5.482248] ath: EEPROM indicates we should expect a direct regpair map
[    5.482249] ath: Country alpha2 being used: 00
[    5.482249] ath: Regpair used: 0x60
[    5.489817] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[    5.490102] ieee80211 phy0: Atheros AR9280 Rev:2 mem=0xffffc90000f40000, irq=17
[    5.491543] ath9k 0000:03:00.0 wlp3s0: renamed from wlan0
```

linrunner, 2016-11-28:
See [extracting the image](https://wiki.archlinux.org/index.php?title=Mkinitcpio&redirect=no#Extracting_the_image).

rafwes, 2016-11-29:
You were right, ath9k was not in initramfs. 
This is the image created with no modules specified in /etc/mkinitcpio.conf (standard configuration)

```
rafwes@thinkpad ~ % lsinitcpio -a /boot/initramfs-linux.img    
/usr/bin/lsinitcpio: line 78: warning: command substitution: ignored null byte in input
/usr/bin/lsinitcpio: line 90: warning: command substitution: ignored null byte in input
==> Image: /boot/initramfs-linux.img 
==> Created with mkinitcpio 21
==> Kernel: 4.8.10-1-ARCH
==> Size: 6.12 MiB
==> Compressed with: gzip
  -> Uncompressed size: 15.84 MiB (.386 ratio)
  -> Estimated extraction time: 0,189s

==> Included modules:
  ahci			  led-class		  sdhci-pci
  atkbd			  libahci		  sd_mod
  btrfs			  libata		  serio
  cdrom			  libcrc32c		  serio_raw
  crc32c_generic	  libps2		  sr_mod
  crc32c-intel		  mmc_block		  usb-common
  ehci-hcd		  mmc_core		  usbcore
  ehci-pci		  raid6_pq		  usbhid
  hid			  scsi_mod		  usb-storage
  i8042			  sdhci			  xor

==> Included binaries:
  blkid			  kmod			  switch_root
  busybox		  modprobe		  systemctl
  fsck			  mount			  systemd-tmpfiles
  fsck.btrfs		  sulogin		  udevadm

```

linrunner, 2016-11-29:
What happens when you include ath9k in your initramfs?

rafwes, 2016-11-29:
The problem is no longer present if I bake ath9k into the initramfs. That is not a general fix, however. I suspect something should be changed in tld's systemd unit files to avoid dependency errors.

linrunner, 2016-11-29:
Do you have suggestions on what to change?

[tlp.service](https://github.com/linrunner/TLP/blob/master/tlp.service.in) is already one of the last services started – after multi-user.target. I don't understand why the ath9k interface isn't there at that point in time.

rafwes, 2016-11-30:
I tried messing with the service file to no avail. This might have something to do with udev, but I am very clueless on how to continue to debug this.

linrunner, 2016-11-30:
Could you attach

    system-analyze plot > boot.svg

as file? Created with the original tlp.service of course.

rafwes, 2016-12-02:
There you go:
[boot.zip](https://github.com/linrunner/TLP/files/626959/boot.zip)

linrunner, 2016-12-07:
Dependencies `After=multi-user.target ... NetworkManager.service`work as designed. I have no idea why the interface is not ready when tlp.service is called.

dywisor, 2016-12-07:
@rafwes, you could try waiting until all (boot-time known) udev events have been processed:

```sh
mkdir -p -- /etc/systemd/system/tlp.service.d

cat << EOF > /etc/systemd/system/tlp.service.d/wait-for-udev-settle.conf
[Unit]
Wants=systemd-udev-settle.service
After=systemd-udev-settle.service
EOF
```
Reboot and see if that works.

If it doesn't, revert with ``rm /etc/systemd/system/tlp.service.d/wait-for-udev-settle.conf``.

rafwes, 2016-12-09:
Dwisor, thanks for the pointer. I tried your service file but, powersave is still not being set on boot,.even though the interface seems to come up before tlp now due to the udev settle dependency.

[tlp-stat_udev.txt](https://github.com/linrunner/TLP/files/642650/tlp-stat_udev.txt)
[boot_udev.zip](https://github.com/linrunner/TLP/files/642652/boot_udev.zip)

Very strange.

dywisor, 2016-12-09:
Guess I found the culprit.

From your tlp-stat:
```
-- Reboot --
dez 09 13:58:47 thinkpad tlp[374]: +++ mode=auto (0.9)                                    # udev #1
dez 09 13:58:47 thinkpad tlp[375]: PATH=/usr/local/sbin:/usr/local/bin:/usr/bin           # udev #1
dez 09 13:58:47 thinkpad tlp[378]: power_source=bat                                       # udev #1
dez 09 13:58:47 thinkpad tlp[383]: +++ mode=auto (0.9)                                    # udev #2
dez 09 13:58:47 thinkpad tlp[385]: PATH=/usr/local/sbin:/usr/local/bin:/usr/bin           # udev #2
dez 09 13:58:47 thinkpad tlp[387]: power_source=bat                                       # udev #2
dez 09 13:58:47 thinkpad tlp[388]: lock_tlp_nb.locked                                     # udev #1 (or #2)
dez 09 13:58:47 thinkpad tlp[392]: compare_and_save_power_state(1).different: old=        # udev #1 (or #2)
dez 09 13:58:47 thinkpad tlp[394]: lock_tlp_nb.failed                                     # udev #2 (or #1)
...
dez 09 13:58:47 thinkpad tlp[433]: set_wifi_power_mode(1).get_wifi_ifaces: wifaces=       # udev #1 (or #2)
...
dez 09 13:58:49 thinkpad tlp[592]: +++ mode=init start (0.9)                              # init
...
dez 09 13:58:49 thinkpad tlp[637]: compare_and_save_power_state(1).equal                  # init
```

There are two power supply related udev events during bootup that cause TLP to run its on-power-source-change code ("tlp auto") before tlp.service is started. At that point in time, no wireless/wired interfaces are present yet, hence the "no wifaces" log entry. 

I'd assume that the wireless/wired interfaces are ready when tlp.service kicks in, but "tlp init start" detects that there have been no power source changes since the last tlp auto/init run and therefore wifi powersave settings are not applied (again).

Please verify by changing line 114 in ``/usr/bin/tlp`` as follows:
```diff
-                  if compare_and_save_power_state $pwrmode; then
+                  if compare_and_save_power_state $pwrmode || true; then
```

rafwes, 2016-12-10:
@dywisor, you found it! Bug solved. Thanks!

@linrunner please mark this as solved. Thank you too.

linrunner, 2016-12-10:
@dywisor: brilliant analysis! 

Must disable the check in L114 completely ... or do you have some more ideas?

closed 2016-12-11 by linrunner