linrunner/TLP #239: Bluetooth radio cannot detect visible bluetooth devices with TLP installed

mokalovesoulmate, 2016-12-06:
In my case this is a bluetooth mouse (Microsoft Sculpt Comfort).

Kernel version: Linux ThinkPad-E450 4.8.0-22-generic #24-Ubuntu SMP Sat Oct 8 09:15:00 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

PC: Lenovo Thinkpad E450

### Symptom data

1. Does the problem occur on battery or AC or both?
Both.

2. Attach the *full* output of `tlp-stat` via [Gist](https://gist.github.com/) for *all* cases of 1.
[will edit once I reproduce the problem, at the moment this problem is fixed by uninstall tlp then reinstall linux kernel then reboot]

### Expected behavior
Bluetooth mouse still connected despite tlp activation

### Actual behavior
Computer no longer detect or connect to any of bluetooth devices visible

### Steps to reproduce the problem
1) Fresh ubuntu install (optional)
2) Connect bluetooth mouse
3) Install tlp (additional thinkpad is optional)
4) Activate tlp

### Temporary fix
1) Uninstall (purge) tlp and tlp-rdw:
sudo apt-get remove --purge tlp tlp-rdw

2) Reinstall linux kernel:
sudo apt-get install --reinstall linux-image-$(uname -r)

3) Reboot computer.

linrunner, 2016-12-06:
Hi,

try the workaround from the [FAQ](http://linrunner.de/en/tlp/docs/tlp-faq.html#input).

mokalovesoulmate, 2016-12-07:
It works! Thanks!

Note: sometimes bluetooth settings not working because "Bluetooth is disabled" but you can turn on, off, and on again.

closed 2016-12-07 by mokalovesoulmate

SaulEfrain, 2017-09-18:
Hello, the solution might be found here: http://linrunner.de/en/tlp/docs/tlp-faq.html  ...It worked for me.

Configure TLP:
USB_BLACKLIST_BTUSB=1