linrunner/TLP #241: acpi_call not installed in kernel

madgoks, 2016-12-24:
### Symptom data

1. The problem occurs on both AC as well as battery.
2. Output of tlp-stat on battery  and AC power is attached as two separate files.

### Expected behavior
After having installed akmod-acpi_call from the testing repo for Fedora 25,pacpi_bat should not show up as inactive. 

### Actual behavior
After having installed akmod-acpi_call from the testing repo for Fedora 25, it shows up as inactive. Also if one does a modprobe, it does not show up in the kernel `modules.
`modprobe -v acpi_call`
_modprobe: FATAL: Module acpi_call not found in directory /lib/modules/4.8.14-300.fc25.x86_64

[tlp_status_batt.txt](https://github.com/linrunner/TLP/files/671707/tlp_status_batt.txt)
[tlp_status_ac.txt](https://github.com/linrunner/TLP/files/671708/tlp_status_ac.txt)
_

### Steps to reproduce the problem
I followed all the installation steps given on the website but couldnt resolve the issue.

adamczyk777, 2017-05-25:
Same issue here, happens for new kernel versions all the time

ValdikSS, 2017-09-02:
acpi_call can't be compiled automatically since akmods build it with `-Werror`, which treats all warnings as errors.
```
  CC [M]  /tmp/akmodsbuild.PnJ18ZSt/BUILD/acpi_call-kmod-1.1.0/_kmod_build_4.12.9-200.fc25.x86_64/acpi_call.o
/tmp/akmodsbuild.PnJ18ZSt/BUILD/acpi_call-kmod-1.1.0/_kmod_build_4.12.9-200.fc25.x86_64/acpi_call.c: In function 'acpi_proc_write':
/tmp/akmodsbuild.PnJ18ZSt/BUILD/acpi_call-kmod-1.1.0/_kmod_build_4.12.9-200.fc25.x86_64/acpi_call.c:278:9: error: implicit declaration of function 
     if (copy_from_user( input, buff, len )) {
         ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
```

@linrunner can you update acpi_call in Fedora 25 repository please?

ValdikSS, 2017-09-02:
@linrunner you need to add the following line into acpi_call.c:
`#include <linux/uaccess.h>`

SiliconKite, 2018-01-05:
Same problem here, any solutions ? Thanks

closed 2018-02-18 by linrunner

peculiarity, 2019-06-11:
Why was this issue closed? I still experience the following issue with Fedora 30