linrunner/TLP #242: Slow shutdown

Mystro256, 2016-12-31:
A Fedora user has reported this issue:

    Description of problem:
    When TLP installed, Fedora 25 shutdown slower

    Version-Release number of selected component (if applicable):
    TLP 0.9

    How reproducible:
    To try: systemctl stop tlp
    It will be very slow
    Run : ps 
    We can see, nmcli process hanging

Output from tlp-stat:
https://paste.fedoraproject.org/513628/83564914/

Mirror link if that one is unavailable:
https://dl.dropboxusercontent.com/u/42480493/1408434-tlp-stat.txt

This issue occurs on both battery and AC, and a follow-up comment was given by the reporter:

    Found workaround by commenting this line in config:
    DEVICES_TO_DISABLE_ON_LAN_CONNECT="wifi wwan"

See original bugreport:
https://bugzilla.redhat.com/show_bug.cgi?id=1408434

linrunner, 2017-01-04:
Hi,

am i right to assume the delay occurs when the lan cable is connected and thus wifi is off? In this state TLP will turn _on_ wifi during shutdown. 

Please try if the delay occurs with:

   wifi on

biji, 2017-01-18:
Hi..
with DEVICES_TO_DISABLE_ON_LAN_CONNECT="wifi wwan" in config
cable connected, then wifi off automatically

I ran: wifi on -> fast

Then I simulate shutdown, by running systemctl stop tlp -> 20s

linrunner, 2017-01-19:
[Enable trace mode](http://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html#trace) and show the trace output for

    systemctl stop tlp 

Omit everything with a timestamp before the command was issued.

biji, 2017-01-20:
`
Jan 20 22:26:37 onyx.local tlp[10194]: +++ mode=init stop (0.9) ++++++++++++++++++++++++++++++++++++++++
Jan 20 22:26:37 onyx.local tlp[10195]: PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin
Jan 20 22:26:37 onyx.local tlp[10196]: power_source=bat
Jan 20 22:26:37 onyx.local tlp[10199]: lock_tlp.locked
Jan 20 22:26:37 onyx.local tlp[10200]: set_radio_device_states(stop): enable=wifi  disable=
Jan 20 22:26:37 onyx.local tlp[10201]: device_switch(wifi, on, , )
Jan 20 22:26:37 onyx.local tlp[10204]: get_devc(wifi) = /sys/class/rfkill/rfkill1/state
Jan 20 22:26:37 onyx.local tlp[10206]: get_devs(wifi) = 1
Jan 20 22:26:37 onyx.local tlp[10207]: device_switch(wifi, on).desired_state
Jan 20 22:26:37 onyx.local tlp[10210]: invoke_nmcli(wifi, on).nm: rc=2
Jan 20 22:27:02 onyx.local tlp[10232]: invoke_nmcli(wifi, on).radio: rc=1
Jan 20 22:27:02 onyx.local tlp[10234]: get_devs(wifi) = 1
Jan 20 22:27:02 onyx.local tlp[10235]: device_switch(wifi, on).no_change: rc=4
Jan 20 22:27:02 onyx.local tlp[10181]: Enabling radios: wifi.
Jan 20 22:27:02 onyx.local tlp[10241]: set_laptopmode(0): 0
Jan 20 22:27:02 onyx.local tlp[10242]: set_dirty_parms(0): 1500
Jan 20 22:27:02 onyx.local tlp[10243]: set_scaling_governor(0): powersave
Jan 20 22:27:02 onyx.local tlp[10244]: set_cpu_perf_pct(0).min: 20
Jan 20 22:27:02 onyx.local tlp[10245]: set_cpu_perf_pct(0).max: 100
Jan 20 22:27:02 onyx.local tlp[10246]: set_cpu_boost_all(0).intel_pstate: 1
Jan 20 22:27:02 onyx.local tlp[10247]: set_sched_powersave(0).not_available
Jan 20 22:27:02 onyx.local tlp[10248]: set_nmi_watchdog: 0
Jan 20 22:27:02 onyx.local tlp[10249]: set_phc_controls.not_configured
Jan 20 22:27:02 onyx.local tlp[10252]: set_energy_perf_policy(0): performance
Jan 20 22:27:02 onyx.local tlp[10259]: set_disk_apm_level(0): sda [] 254
Jan 20 22:27:02 onyx.local tlp[10264]: set_disk_io_sched: sda [] cfq
Jan 20 22:27:02 onyx.local tlp[10265]: set_sata_link_power(0).max_performance: /sys/class/scsi_host/host0
Jan 20 22:27:02 onyx.local tlp[10266]: set_sata_link_power(0).max_performance: /sys/class/scsi_host/host1
Jan 20 22:27:02 onyx.local tlp[10267]: set_sata_link_power(0).max_performance: /sys/class/scsi_host/host2
Jan 20 22:27:02 onyx.local tlp[10268]: set_ahci_runtime_pm(0).not_configured
Jan 20 22:27:02 onyx.local tlp[10269]: set_pcie_aspm(0).disabled_by_kernel
Jan 20 22:27:02 onyx.local tlp[10270]: set_radeon_profile(0).no_card
Jan 20 22:27:02 onyx.local tlp[10275]: set_wifi_power_mode(0, wlp4s0).iw: off; rc=0
Jan 20 22:27:02 onyx.local tlp[10278]: disable_wake_on_lan: enp0s25
Jan 20 22:27:02 onyx.local tlp[10280]: set_sound_power_mode(0).hda: 0 controller=Y
Jan 20 22:27:02 onyx.local tlp[10282]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:00.0 [0x060000 Host Bridge]
Jan 20 22:27:02 onyx.local tlp[10284]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:02.0 [0x030000]
Jan 20 22:27:02 onyx.local tlp[10286]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:03.0 [0x040300 Audio device]
Jan 20 22:27:02 onyx.local tlp[10288]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:14.0 [0x0c0330]
Jan 20 22:27:02 onyx.local tlp[10290]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:16.0 [0x078000]
Jan 20 22:27:02 onyx.local tlp[10292]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:19.0 [0x020000 Ethernet controller]
Jan 20 22:27:02 onyx.local tlp[10294]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:1b.0 [0x040300 Audio device]
Jan 20 22:27:02 onyx.local tlp[10296]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:1c.0 [0x060400]
Jan 20 22:27:02 onyx.local tlp[10298]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:1c.2 [0x060400]
Jan 20 22:27:03 onyx.local tlp[10300]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:1c.4 [0x060400]
Jan 20 22:27:03 onyx.local tlp[10302]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:1c.5 [0x060400]
Jan 20 22:27:03 onyx.local tlp[10304]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:1d.0 [0x0c0320]
Jan 20 22:27:03 onyx.local tlp[10306]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:1f.0 [0x060100]
Jan 20 22:27:03 onyx.local tlp[10308]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:1f.2 [0x010601]
Jan 20 22:27:03 onyx.local tlp[10310]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:1f.3 [0x0c0500]
Jan 20 22:27:03 onyx.local tlp[10312]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:1f.6 [0x118000]
Jan 20 22:27:03 onyx.local tlp[10314]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:04:00.0 [0x028000 Wireless]
Jan 20 22:27:03 onyx.local tlp[10316]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:05:00.0 [0x038000]
Jan 20 22:27:03 onyx.local tlp[10318]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:06:00.0 [0xff0000]
Jan 20 22:27:03 onyx.local tlp[10181]: Applying power save settings...done.
Jan 20 22:27:03 onyx.local tlp[10319]: unlock_tlp

`

linrunner, 2017-01-22:
This seems a bit weird to me. The delay happens when TLP issues the command `nmcli radio wifi on` as can be seen here

> Jan 20 22:26:37 onyx.local tlp[10206]: get_devs(wifi) = 1
> Jan 20 22:26:37 onyx.local tlp[10207]: device_switch(wifi, on).desired_state
> Jan 20 **22:26:37** onyx.local tlp[10210]: invoke_nmcli(wifi, [...]

But `systemctl stop tlp` (or `tlp init stop`) calls exactly the same code for wifi as `wifi on` does.

Can you show the output of

    nmcli radio

before and after both commands? 

Before each test please disconnect/reconnect the LAN cable briefly to have a defined state.

biji, 2017-01-23:
**systemctl stop tlp**

before command: 

WIFI-HW  WIFI      WWAN-HW  WWAN    
enabled  disabled  enabled  enabled 

after: 

WIFI-HW  WIFI      WWAN-HW  WWAN    
enabled  disabled  enabled  enabled 

**tlp init stop**

before command: 

WIFI-HW  WIFI      WWAN-HW  WWAN    
enabled  disabled  enabled  enabled 

after:

WIFI-HW  WIFI     WWAN-HW  WWAN    
enabled  enabled  enabled  enabled

linrunner, 2017-01-26:
So the same code – `tlp init stop` – works when invoked directly but stalls and fails when invoked by systemd (NM wifi stays disabled).

linrunner, 2017-02-09:
Status update: i can reproduce the issue here on F25.

gion86, 2017-03-17:
I have this bug too on Fedora 25. Is there any way to bypass this until you fix it?
Or can I help in any way?

linrunner, 2017-03-17:
Depends on your configuration.

gion86, 2017-03-18:
Sorry...  I've seen now on the first post here: commenting out DEVICES_TO_DISABLE_ON_LAN_CONNECT works...

linrunner, 2017-03-19:
Workaround [committed](https://github.com/linrunner/TLP/commit/ef6890bbdaf4059c71d1fcb52ec7c8d54bfdbcbe). 

[Patch for 0.9](https://github.com/linrunner/TLP/commit/dabd0e900f6e10c358436c584c51dd1f55c320d2) too.

Mystro256, 2017-03-21:
@gion86 I've pushed the workaround to updates-testing, please test and leave some feedback if you can:
https://bodhi.fedoraproject.org/updates/FEDORA-2017-90768381ad

gion86, 2017-03-23:
I've tested it and it works, the shutdown time is normal.
But from the stop command:
`systemctl stop tlp`

I see the status of the tlp service is **failed**, and the log says:

> mar 22 19:24:16 latE7270 tlp[6153]: Disabling radios: bluetooth.
> mar 22 19:24:41 latE7270 tlp[6153]: Enabling radios: wifi.
> mar 22 19:24:41 latE7270 systemd[1]: Started TLP system startup/shutdown.
> mar 22 19:33:35 latE7270 systemd[1]: Stopping TLP system startup/shutdown...
> mar 22 19:33:35 latE7270 tlp[7957]: Applying power save settings...done.
> mar 22 19:33:38 latE7270 systemd[1]: tlp.service: Stopping timed out. Terminating.
> mar 22 19:33:38 latE7270 tlp[7957]: Enabling radios: wifi
> mar 22 19:33:38 latE7270 systemd[1]: Stopped TLP system startup/shutdown.
> mar 22 19:33:38 latE7270 systemd[1]: tlp.service: Unit entered failed state.
> mar 22 19:33:38 latE7270 systemd[1]: tlp.service: Failed with result 'timeout'

So it's a timeout, is that normal?

linrunner, 2017-03-24:
That's right.

gion86, 2017-03-25:
OK, but it's a temporary fix or not? In any case I've tried other shutdown and it does not hang..

Mystro256, 2017-03-28:
This is just a workaround for the issue, so I would think it's temporary until the root cause can be determined.

linrunner, 2017-04-17:
The cause for this issue is an incomplete SELinux policy module in Fedora. I added an writeup and AVC data to the [related Fedora bug report](https://bugzilla.redhat.com/show_bug.cgi?id=1403782#c6).

biji, 2017-05-10:
FYI my tlp policy : https://pastebin.com/SnF9C3bQ

linrunner, 2018-01-24:
Fixed in Fedora's package **selinux-policy** --> https://bugzilla.redhat.com/show_bug.cgi?id=1403782

closed 2018-01-24 by linrunner