linrunner/TLP #243: Configure bayoff on AC

toonn, 2017-01-03:
This is a feature request not a bug. I apologize if this isn't the appropriate place for this but there doesn't seem to be any other official forum or channel to talk about TLP.

I rarely use the optical disk drive so there's no point in having it switch between on and off every time I connect AC power. I'd like to be able to configure the ODD to always be off.

linrunner, 2017-01-03:
Why don't you just put 

    tlp bayoff

in your *rc.local* (or similar)?

toonn, 2017-01-03:
I don't like spreading around configuration like this. I'll forget if I ever remove tlp or do a reinstall of the system. Why is it restricted to battery mode in TLP in the first place? It'd be nice to have all the power related config in one place, the TLP configuration.

linrunner, 2017-01-03:
Please show me the full output of 

    tlp-stat

via [Gist](http://gist.github.com/).

toonn, 2017-01-03:
Here you go: https://gist.github.com/toonn/6a14b616b44431c0c728898639e220f1

linrunner, 2017-01-03:
OK. I'll consider it for the next release.

toonn, 2017-01-03:
Thank you. TLP is great btw.

linrunner, 2017-01-11:
Done.

toonn, 2017-01-12:
One more thing. Commenting here because I'm not sure whether it's an actual issue.
The ultrabay powers back on after suspending (at least when plugged in), is this a bug or is there no way around this?

linrunner, 2017-01-12:
Your question is about the development version (master branch) or 0.9?

Expected behaviour is that the off state is preserved over the suspend/resume cycle. Make shure **tlp-sleep.service** is enabled (see [FAQ](http://linrunner.de/en/tlp/docs/tlp-faq.html#systemdunits)). `tlp-stat -s` will issue a warning if not.

linrunner, 2017-01-15:
Please download shiny new packages for Debian/Ubuntu [here](http://download.linrunner.de/packages/), test your feature request and report back. Thank you.

toonn, 2017-01-16:
Ok, I changed the config which now works as intended when on battery.
Then I installed 0.9.901, not the rdw package because I did not install that previously.
Do I need to add some configuration for this to work? On battery it still works as intended but after booting on ac the ultrabay stays on and after waking from suspend the ultrabay turns on if it was off.
Note that I am on ubuntu 15.10.

linrunner, 2017-01-16:
For power off on ac you'll need **BAY_POWEROFF_ON_AC=1** of course. Refer to https://github.com/linrunner/TLP/blob/master/default#L169

I suggest you copy the config file from /etc/default/tlp.dpkg-dist to tlp and add your desired changes.

> Note that I am on ubuntu 15.10.

Sorry, i won't provide update packages for an EOL Ubuntu release via the PPA.

toonn, 2017-01-17:
Ok, seems to work. Ultrabay turns off after boot and switches back on, then off after waking from suspend.
Thanks for this!

I am having some trouble with the laptop shutting down without warning when running out of battery though. I don't think this is because of the new version but I was wondering whether this could be because of TLP?
I'm guessing not but I don't know what else the problem could be. Could it be the battery misreporting it's charge?

linrunner, 2017-01-19:
> and switches back on, then off after waking from suspend.

This must happen too when you connect the power supply in suspended state, even with BAY_POWEROFF_ON_AC=**0**.

linrunner, 2017-05-25:
Released with 1.0.

closed 2017-05-25 by linrunner