linrunner/TLP #244: TLP conflicts with Bumblebee, disabling GPU until reboot

jkdf2, 2017-01-05:
### Symptom data

1. Does the problem occur on battery or AC or both?

After going to battery, even if AC reconnected, until reboot.

2. Attach the *full* output of `tlp-stat` via [Gist](https://gist.github.com/) for *all* cases of 1.

https://gist.github.com/jkdf2/c296e7dc8cf69f94efa517a1eb2b7fc0

### Expected behavior

`optirun glxgears` should be able to be started when on battery and especially when on AC - it's a deliberate user choice.

### Actual behavior

After being on battery-only, `optirun glxgears` reports: 

`[  525.648588] [ERROR]Cannot access secondary GPU - error: Could not enable discrete graphics card`

Unless `RUNTIME_PM_BLACKLIST` and `RUNTIME_PM_DRIVER_BLACKLIST` are configured correctly. Perhaps this can be automated on TLP install if it could detect an Nvidia Optimus configuration?

### Steps to reproduce the problem

Have a device that supports Nvidia Optimus, the technology that uses the integrated graphics but switches to the discrete graphics card as required, and install the [Bumblebee driver](https://wiki.archlinux.org/index.php/bumblebee) to manage it. Then attempt to run an application in this mode with `optirun`.

For reference, see [this GitHub issue over at Bumblebee](https://github.com/Bumblebee-Project/Bumblebee/issues/829).

linrunner, 2017-01-05:
> Perhaps this can be automated on TLP install if it could detect an Nvidia Optimus configuration?

I see this as a feature request. If you want to have it, send me a pull request :-).

ps. i don't own optimus hardware (and never will).

jkdf2, 2017-01-05:
Thanks @linrunner.

Anybody who comes across this can assume this is open for development. I don't have time at the moment to make this a priority, unless I follow-up otherwise.

ArchangeGabriel, 2017-01-07:
Reliably detect: search for two VGA+3D devices in PCI devices (so VGA+VGA, VGA+3D or 3D+3D, though I’ve never seen that one AFAIR), one being Intel and the other Nvidia (you want to avoid Nvidia+Nvidia for instance). The PCI ID is definitively a bad idea, we’ve seen it going from 01 to 10 at least.

However, the issue is to be (partially?) fixed in bbswitch, since supporting 4.8 kernel PCIe port PM require making bbswitch a proper driver. This is the target for next milestone, but only @Lekensteyn is working on it and he has been quite busy for a while.

Lekensteyn, 2017-01-07:
In my udev I have this rule to match nvidia GPUs:
```
# 2016-10-23 let nouveau detect the device and power off as appropriate
SUBSYSTEM=="pci", ATTR{vendor}=="0x10de", ATTR{class}=="0x030000", GOTO="powersave_end"
```

Starting with kernel 4.8, PCIe ports support RPM, but this may have undesired effects if no PCI driver is loaded for the Nvidia GPU (on an hybrid graphics laptop). The result is that loading the driver fails, unless you remove the PCI device via sysfs and rescan the parent PCIe bus. No idea why that is the case though.

Maybe you could do a similar thing, disabling RPM for all nvidia GPUs as above. When nouveau is in use, it will enable RPM itself. When nvidia blob is in use, RPM is not usable anyway.

linrunner, 2017-01-08:
@jkdf2 : let's see if your card fits @Lekensteyn 's rule? 

    lspci -vnn -s 01:00.0

ArchangeGabriel, 2017-01-08:
`ATTR{vendor}=="0x10de"` for sure, but `ATTR{class}=="0x030000"` I don’t think so, because this is for `VGA compatible controller [0300]`, but as I’ve stated above you’re also going to have `3D controller [0302]`, which is probably `ATTR{class}=="0x030200"`.

Here is mine:
```
02:00.0 3D controller [0302]: NVIDIA Corporation GK107M [GeForce GT 750M] [10de:0fe4]
```

That’s being said, @Lekensteyn idea to blacklist all Nvidia GPU from RPM looks fine indeed.

jkdf2, 2017-01-08:
`lspci -vnn -s 01:00.0` output:

```
01:00.0 3D controller [0302]: NVIDIA Corporation GM107M [GeForce GTX 960M] [10de:139b] (rev ff) (prog-if ff)
	!!! Unknown header type 7f
	Kernel modules: nvidia
```

ArchangeGabriel, 2017-01-08:
@jkdf2 (and @linrunner for info) If you want the full output, you have to power on the card first.

```
sudo tee /proc/acpi/bbswitch <<< ON
lspci -vnn -s 01:00.0
sudo tee /proc/acpi/bbswitch <<< OFF
```

linrunner, 2017-01-08:
It's fine like that, because detection must work even when the card is off.

ArchangeGabriel, 2017-01-08:
Sure. :)

Lekensteyn, 2017-01-09:
The above udev rule works for me:

    01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GM204M [GeForce GTX 965M] [10de:13d9] (rev a1)

For more compatibility you might want to include both the 0300 and 0302 classes.

linrunner, 2017-01-15:
Please download shiny new packages for Debian/Ubuntu [here](http://download.linrunner.de/packages/), test your feature request and report back. Thank you.

Arch Linux users: use tlp-git and tlp-rdw-git from the AUR.

jkdf2, 2017-01-15:
I admit that I don't have the understanding of RPM that y'all do but it seems to be working great for me. I purged `tlp` and `tlp-rdw` and installed the beta version.

`optirun glxgears` now works correctly:
* On AC
* On battery after disconnecting AC
* After suspend, on battery

Here's an output of `tlp-stat` in case it helps:

https://gist.github.com/jkdf2/bdb8262911d369655a55855701181c07

Thanks! 🎆

linrunner, 2017-01-19:
Fine, thanks for testing.

ion-storm, 2017-02-02:
can confirm the beta fixed compatibility with bumblebee, thanks for the fix!

SalahAdDin, 2017-05-12:
I can confirm this error also.

SalahAdDin, 2017-05-12:
I'll applied the provisional solution.

linrunner, 2017-05-25:
Released with 1.0.

closed 2017-05-25 by linrunner

overclockedllama, 2018-05-21:
can this patch be applied to tlp-pmu as i would like to be able to use pm-utils as well

MarkoSchuetz, 2019-02-04:
I have TLP installed, but I currently run the nvidia driver only. In this setup I noticed an enormous battery drain during suspend to RAM (79% drain over 16 hours). While I'm not sure that this was the cause, I suspect it was due to 
`RUNTIME_PM_DRIVER_BLACKLIST="amdgpu nouveau nvidia radeon pcieport"`
from which I have since removed `nvidia`.
Maybe `/etc/defaults/tlp` should check whether bumblebee is in use and blacklist drivers it controls only in that case...