linrunner/TLP #245: Documentation for DESTDIR in Makefile

jkdf2, 2017-01-05:
Hi! The [Developer Documentation](http://linrunner.de/en/tlp/docs/tlp-developer-documentation.html) on the website is extremely helpful.

However, I am missing one point: how to build into my current directory to test without affecting my system. I looked in the Makefile and I see there is a `$(DESTDIR)` parameter. But `make DESTDIR=built/ all` does not place the output files in `built/`. Could you provide a quick mention of how to use this in the documentation? Thanks so much 😄

linrunner, 2017-01-05:
Your command works fine here, even when the target directory doesn't exist. I suggest you show the actual output of it.

jkdf2, 2017-01-05:
Sorry, I mean it does work, there's just no note of it in the Documentation.

linrunner, 2017-01-05:
No anymore.

linrunner, 2017-01-05:
ps. i didn't mention it explicitly based on the assumption that every OSS Makefile has it ;-)

closed 2017-01-05 by linrunner

linrunner, 2017-01-05:
ps2. For my own testing workflow i never use a plain **make install**. Instead i always build a package. If you run Debian/Ubuntu it's quite simple to obtain the current package files:

    git checkout debian/current -- debian
    git rm -r --cached debian

then create the packages with

    dpkg-buildpackage -b

For other distros get the rpm specfile or PKGBUILD or whatever.