linrunner/TLP #248: Init script issues

Mystro256, 2017-01-17:
When packaging TLP for epel6, which uses the old init system instead of systemd, the following warnings are generated by rpmlint:

    no-chkconfig-line:
    The init script doesn't contain a chkconfig line to specify the runlevels at
    which to start and stop it.

    subsys-not-used:
    While your daemon is running, you have to put a lock file in
    /var/lock/subsys/. To see an example, look at this directory on your machine
    and examine the corresponding init scripts.

I'm not as familiar with how init scripts work, so I'm not sure if these should be ignored or not, but I would assume the runlevel/chkconfig should probably be set and the subsys should be ignored (unless it's valuable to you).

linrunner, 2017-01-17:
> no-chkconfig-line

I suggest to add the line

    # chkconfig: 2345 98 01

to the init script header. I'll add it upstream with 1.0.

> subsys-not-used:

There's no daemon and `tlp init` itself is stateless, so a lock file doesn't make sense. My quick search didn't discover a a way to suppress this. Ignoring the warning seems the best solution.

I found the information [here](http://fedoraproject.org/wiki/EPEL:SysVInitScript) helpful.

Mystro256, 2017-01-17:
Sounds good, I thought as much.

Thanks for your help.

Mystro256, 2017-01-17:
As another request, can you add a variable to specify the initdir in the makefile. On EPEL it's /etc/rc.d/init.d instead of /etc/init.d/tlp, so it would be easier to change it via a variable than using a patch.

linrunner, 2017-01-17:
OK.

linrunner, 2017-01-18:
Done.

closed 2017-01-18 by linrunner