linrunner/TLP #249: Feature request: HWP power hint

ishitatsuyuki, 2017-01-22:
According to the [docs](https://www.kernel.org/doc/Documentation/cpu-freq/intel-pstate.txt) of linux@master, HWP power hint support via `energy_performance_preference` is coming in 4.10.

I would be glad to see that also supported in TLP.

linrunner, 2017-01-22:
From the docs i understand the attributes are

    /sys/devices/system/cpu/cpu*/cpufreq/energy_performance_available_preferences
    /sys/devices/system/cpu/cpu*/cpufreq/energy_performance_preference

Right?

ya-isakov, 2017-03-01:
I'm not original reporter, but I also wants this feature. Yes, the files are correct. First one is only for displaying available policies (and for now, they're hard-coded). If I understand correctly, this option should override x86_energy_perf_policy, but in 4.10 it looks like they're independent,

ishitatsuyuki, 2017-03-10:
Here's a ping, any updates?

linrunner, 2017-03-21:
If you're in a hurry, you may send patches anytime ;-))

Please test if the setting applies and show the output of 

    tlp-stat -p

I don't own a Skylake laptop.

ya-isakov, 2017-03-21:
Not working for me - path should be
/sys/devices/system/cpu/cpu*/cpufreq/energy_performance_preference
instead of
/sys/devices/system/cpu/cpu*/energy_performance_preference

linrunner, 2017-03-21:
Path fixed: https://github.com/linrunner/TLP/commit/e357cdc9f66ea47c5621682ff75984dab38df100

What about tlp-stat?

ya-isakov, 2017-03-22:
Now everything works properly, thank you!

ishitatsuyuki, 2017-03-22:
Closing as resolved. I'm not sure how effective this hint is, though.

closed 2017-03-22 by ishitatsuyuki

linrunner, 2017-03-22:
Thank you for testing. 

Could you both please post the output of

    tlp -s -c -p

ya-isakov, 2017-03-22:
Do yo mean tlp-stat... ? Here is mine https://nopaste.me/view/d583d8d1

linrunner, 2017-03-22:
Indeed, thanks.

May I ask why you didn't enable TLP's service units? See [FAQ](http://linrunner.de/en/tlp/docs/tlp-faq.html#system).

ya-isakov, 2017-03-22:
I had no chance to check what they need for - everything works without them, it seems, so I just kept them not enabled. Maybe I should check now...

linrunner, 2017-03-22:
> everything works without them, it seems

Wrong. See FAQ link above.

ya-isakov, 2017-03-22:
I've checked suspend/resume - even without sleep.service, no difference in tlp-stat output before suspend and after resume. For radio devices - I'm never disabling wi-fi, and gnome desktop handles bluetooth disable/enable for me. So, I'll probably enable tlp.service for initial config of polices/etc. Thank you! And BTW, thanks a lot for such a great program!

linrunner, 2017-03-24:
I've enabled HWP by default:

    CPU_HWP_ON_AC=balance_performance
    CPU_HWP_ON_BAT=balance_power

Any objections?