linrunner/TLP #250: Add FAQ: what is the best way to only enable a small part of tlp?

jidanni, 2017-01-22:
Please add a FAQ:

How do I just get the benefits of e.g., "tlp usb",
without enabling any other parts of the package?

(You see they say to use "tlp usb" to fix the fact that
on the console I get these messages about the mouse disconnecting every
62 seconds (checked with
perl -nwle 'next unless /PixArt$/; m/\[\s*(\d+)/ or die; $this=$1; print $this-$last; $last=$this;' /var/log/syslog
) just like https://www.spinics.net/lists/linux-usb/msg88994.html
(See also
https://bbs.archlinux.org/viewtopic.php?id=150114
https://bbs.archlinux.org/viewtopic.php?pid=1426292
https://forums.gentoo.org/viewtopic-p-7907316.html
http://hamwaves.com/usb.autosuspend/en/ )

linrunner, 2017-01-22:
TLP is not modular by design. This would be quite difficult to describe and even more difficult to understand for the common user.

The WWW is full of examples about employing USB autosuspend and every other kernel powersave feature, so i think it's far more efficient you write a little script for your special case.

closed 2017-01-22 by linrunner

jidanni, 2017-01-22:
OK, maybe put the above Q & A in the FAQ.