linrunner/TLP #251: Feature-request: "Gaming mode" (is this the wrong place to put this?)

janicez, 2017-01-26:
Yes I know TLP is all about saving power. I also know you've been staying out of brightness type stuff.

This might count as brightness type stuff, but hear me out.

Some people want their computer to be in an endurance-racing, low-performance mode usually when on battery or AC, unless they're running an application whose demands on the system are non-trivial (e.g. FlightGear Flight Simulator, Blender, Quake, video editors, bench-mark programs) - in which case they want to let their battery run down faster in exchange for much better performance. I propose adding "gaming modes" of the AC and battery modes, only activated if a certain process is running. The implementation will be left up to whoever is stuck with this problem, be it me or anyone else.

Hypothetically, the mode could be configured with a /etc/default/tlp line looking like `GAME_PROCESSES="fgfs kdenlive"` and would duplicate every /etc/default/tlp line specifying AC or BAT modes so that ACGAME and BATGAME can be configured to have higher settings (so rather than an 85% maximum pstate on AC, 100% because you're okay with the fan screaming (yes I use TLP as a noise reduction tool, don't tase me), or rather than 60% and no turbo, 87% and turbo available on battery, because you are willing to trade off battery time). It could also just be a disable-power-management switch - when those programs appear, hard disk is set to no restriction and CPU performance ceiling is set to 100% pstate or maximum frequency.

linrunner, 2017-01-26:
No thanks. But nice try.

closed 2017-01-26 by linrunner

janicez, 2017-01-27:
So I guess I should just pretend I didn't post this.

linrunner, 2017-01-27:
It's the right place for feature requests. I just won't implement this one.