linrunner/TLP #252: Script '/etc/NetworkManager/dispatcher.d/99tlp-rdw-nm' took too long; killing it.

leggewie, 2017-02-09:
Hi,

I wrote you an e-mail about this some time last year.  Here is my proper bug ticket.

I've noticed the following two lines in the syslog of my Thinkpad X200s and thought I should let you know about it.  I do not notice any problems otherwise.

Feb  7 18:37:18 localhost nm-dispatcher.action: Script '/etc/NetworkManager/dispatcher.d/99tlp-rdw-nm' took too long; killing it.
Feb  7 18:37:26 localhost NetworkManager[1185]: <warn> Dispatcher script timed out: Script '/etc/NetworkManager/dispatcher.d/99tlp-rdw-nm' timed out.

leggewie, 2017-02-09:
[tlp-stat.txt](https://github.com/linrunner/TLP/files/764027/tlp-stat.txt)

linrunner, 2017-02-09:
Hi,

i can't see a real malfunction here. Besides you haven't configured any of tlp-rdw's functions the script implements. Might as well be a quirk in NM.

It would be helpful if you could analyze on your own what happens on your system when the script runs. Activating [trace mode](http://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html#trace) and adding more debug output to the [script](https://github.com/linrunner/TLP/blob/master/tlp-rdw-nm.in) may help.

ps. please don't attach output as a file but via http://gist.github.com/

closed 2017-08-22 by linrunner