linrunner/TLP #253: need help with Debian/Ubuntu packaging?

leggewie, 2017-02-09:
I just noticed on http://download.linrunner.de/packages/ that you do not supply the latest release for trusty (which is still what I use).  As Debian Maintainer of a couple of packages I was wondering if you need a hand with the packaging.

linrunner, 2017-02-09:
Hi,

which "latest release" do you mean? ;-)

I suggest you take a look at the [devel docs](http://linrunner.de/en/tlp/docs/tlp-developer-documentation.html#deb). I just added some notes how to build the package yourself.

If you want to help with the [Debian package](https://tracker.debian.org/pkg/tlp), feel free to contact Raphael, the maintainer.

closed 2017-02-09 by linrunner

leggewie, 2017-02-09:
It's kind of funny to ask a question and close the ticket at the same time, don't you think?

Do you want an answer?

linrunner, 2017-02-09:
Closing didn't stop you asking questions either. So go ahead with your answer.

Besides: i do funny things all the time – TLP being among them.