linrunner/TLP #255: tp-smapi-dkms -0.41 error(s?) on Lubuntu 16.04.2

connectionX, 2017-02-20:
Hi, I got a Thinkpad T420 with Lubuntu 16.04.2 Fully updated - and yes i did try to re-install but the error persists and sadly i don't know what to do to give has much feedback has the log does

[gist of crash log of smapi 0.41](https://gist.github.com/connectionX/601e777bb98c2f30a6bf369eefdece00)

Hopefully I'm just an Isolated case and no one else is effected so just in case - reporting this error.

linrunner, 2017-02-20:
Hi,

sorry, wrong place. 

**tp-smapi-dkms** is an official Ubuntu package, so please file a Ubuntu bug report, suggesting to backport [version 0.42-1 from Yakkety](http://packages.ubuntu.com/yakkety/tp-smapi-dkms). 

See the [FAQ](https://localhost/linrunner/tlp/docs/tlp-faq.html#tp-smapi-inst).

connectionX, 2017-02-20:
Sorry for bugging you with an unrelated bug report - I guess I thought "tp? must be related to tlp" I should search more before posting - which is number one reason I don't post at all -  but anyways I just saw there is a on going bug report with some one saying something about the kernel being untested with the dkms  ..anyways I apologize for bothering - also if Anyone else is lost - [link to the bug report](https://bugs.launchpad.net/ubuntu/+source/tp-smapi/+bug/1606447) in case someone lost or in my case confused manages to come by this.

closed 2017-02-20 by connectionX