linrunner/TLP #256: Coding bug in tlp-stat

Mystro256, 2017-02-27:
I have a bug reported upstream fedora:
https://bugzilla.redhat.com/show_bug.cgi?id=1426858

It seems like there is a code path were ```ed``` and ```ef``` are not defined and cause an error at the following lines:
https://github.com/linrunner/TLP/blob/master/tlp-stat.in#L1266
https://github.com/linrunner/TLP/blob/master/tlp-stat.in#L1270

I think this is due to a missing else case where ```$batd/energy_full``` and ```$batd/charge_full``` do not exist. I'm unable to reproduce this myself, so I'm unsure if this is the root cause of the reporter's issue, but I would think this is a possible edge case that should be covered regardless.

linrunner, 2017-02-28:
Hi,

here you are. You may wait for 1.0 or [patch 0.9](https://github.com/linrunner/TLP/commit/9498a85d1cedb78a38b54f243c9c4a3daad5bcc6).

Mystro256, 2017-03-03:
Thanks! looks good to me

linrunner, 2017-05-25:
Released with 1.0.

closed 2017-05-25 by linrunner