linrunner/TLP #258: Ubuntu 16.04 Shutdown Hangs

renebarbosafl, 2017-03-16:
### Symptom data

When using battery and TLP 0.90.1 (installed from PPA) my system is "freezing" at shutdown. The laptop is freezing until I push the power button. 

The final screen is: https://launchpadlibrarian.net/306411830/IMG-20170214-WA0043.jpg

The weird thing is, the problem is ony happening when the system is with some hours of uptime, If i start and then shutdown, it's not occurring. If I run systemctl stop tlp the system is shutting down without issues.

1. Does the problem occur on battery or AC or both?

Only In Battery.

2. Attach the *full* output of `tlp-stat` via [Gist](https://gist.github.com/) for *all* cases of 1.

[stat.txt](https://github.com/linrunner/TLP/files/849094/stat.txt)

### Expected behavior

System shutdown without hangs or delays..

### Actual behavior

Freezing until I push the power button..

### Steps to reproduce the problem

1. Start the system using battery and with TLP 0.90.1 installed;
2. Use it for a few hours;
3. Try to shutdown using GUI or systemctl poweroff command;

linrunner, 2017-03-19:
Hi, 

i suggest you read [TLP Troubleshooting](http://linrunner.de/en/tlp/docs/tlp-troubleshooting.html) and try to isolate the cause.

There's a [FAQ section](http://linrunner.de/en/tlp/docs/tlp-faq.html#system) too.

ps. can't decipher your screenshot :-(.

renebarbosafl, 2017-03-22:
Hello,

I've got it fixed by adding the mei_me  module to the RUNTIME_PM_DRIVER_BLACKLIST config.

Now I can reboot/shutdown properly using battery.

Thanks for the help!

closed 2017-03-22 by renebarbosafl

linrunner, 2017-03-22:
Thanks for reporting your solution.

eRaBxEs, 2017-12-06:
@renebarbosafl , I have a similar issue and to be honest with you, I am a new linux user and I really have no idea of what you mean by "I got it fixed by adding mei_me module to the RUNTIME_PM_DRIVER_BLACKLIST configuration of TLP." I even got to post a question on ask ubuntu with the link below: https://askubuntu.com/questions/983736/my-ubuntu-16-04-lts-does-not-shutdown/983741#983741

I believe that we both are having the same issue, but to do what you just prescribed is what I really do not know? How do I what you did?