linrunner/TLP #259: Printer pauses randomly

mrprobot, 2017-03-21:
### Symptom data

Since Ubuntu 16.04 I had issues with printing. The printer randomly paused and I had to click several times on activate to get him to print. I then noticed that the printer is working fine on a fresh installed system. So I decided to install a fresh Ubuntu and install every single program one after another. After installing tlp and rebooting the issue came back. So I removed it again with:

sudo apt purge tlp tlp-rdw

and the printer is working as expected again. I don't think that this behavior is intended (I know it is mentioned in the FAQ section), because I never had this issue on older Ubuntu versions with tlp installed.

1. Does the problem occur on battery or AC or both?

On battery and AC.

2. Attach the *full* output of `tlp-stat` via [Gist](https://gist.github.com/) for *all* cases of 1.

https://gist.github.com/mrprobot/b8cabe5855e7c62a943bc687f5679e00

### Expected behavior

Printer should always work even with tlp installed.

### Actual behavior

Printer gets randomly paused after installing tlp.

### Steps to reproduce the problem

1) install Ubuntu 16.04 or newer
2) install tlp with: sudo apt install tlp tlp-rdw
3) printer will randomly pause at printing (Samsung ML-1640)

linrunner, 2017-03-21:
Hi,

if the printer is connected via USB (you forgot to tell), try to add it to [USB_BLACKLIST](http://linrunner.de/en/tlp/docs/tlp-configuration.html#usb).

Please refer to the [rroubleshooting guide](http://linrunner.de/en/tlp/docs/tlp-troubleshooting.html) too.

mrprobot, 2017-04-02:
Yes it is connected via USB. I already added the printer to the blacklist but I never had to do this before (on Linux Mint 17 or Ubuntu 14.04) or at least I can't remember. The printer is working fine now and I also posted a "fix" for it [here](https://ubuntu-mate.community/t/printer-pauses-randomly-after-upgrade-to-16-04/5653) because it seems to be a common issue. Thanks for your reply!

closed 2017-04-02 by mrprobot

rimas-kudelis, 2017-06-26:
@linrunner I just had this problem as well: CUPS wasn't detecting my printer, and even when I force-fed it my old config from previous distro, it was unable to print until I added printers device ID to the blacklist.

I think it would be reasonable to allow blacklisting all USB printers automatically, same was WWAN and HID devices are currently blacklisted. Do you think this would be feasible?

rimas-kudelis, 2017-08-08:
@mrprobot a better fix for this issue has just landed as #282.