linrunner/TLP #261: Power consumption really high after suspend

Yutsa, 2017-04-10:
### Symptom data

1. Does the problem occur on battery or AC or both?

It occurs on battery

2. Attach the *full* output of `tlp-stat` via [Gist](https://gist.github.com/) for *all* cases of 1.

This is the [output of `tlp-stat --T`](https://gist.github.com/anonymous/d25eb8de26ecee1b4ad6bd653c43f2a6)

### Expected behavior

The power consumption should stay at the same level before and after the suspend.

### Actual behavior

The power consumption is higher after the suspend because CPU packages can't go to `pc6` sleep state after suspend. It was possible to go to `pc6` before suspend though.

### Steps to reproduce the problem

1. Boot the laptop on battery, check power consumption with `powertop`.

2. Suspend by closing the lid, press power button or explicitly call `systemctl suspend`.

3. Test the power consumption with `powertop` after getting back from suspend state.

I'm including screenshots of the power consumption before and after as well as the sleep states of the CPU.

### Before Suspend

[image]

[image]

### After Suspend

[image]

[image]

linrunner, 2017-04-15:
Please add the output of

    tlp-stat

(1) before suspend and (2) after resume. In battery mode of course.

Yutsa, 2017-04-15:
* [tlp-stat output before suspend](https://gist.github.com/anonymous/de0984fcb3e3d9b99080d2be866dba24)

* [tlp-stat after suspend](https://gist.github.com/Yutsa/fe96038074566f56a1effcb4425cf1cb)

Note that I have the following kernel arguments in my `/etc/default/grub` file : 

```
GRUB_CMDLINE_LINUX_DEFAULT="pcie_aspm=force link_power_management_policy=min_power ath9k.ps_enable=1 snd_hda_intel.power_save=2 snd_hda_intel.power_save_controller=1"
```

And that some powersaving parameters were needed to go to `pc6` states.

linrunner, 2017-04-15:
Your outputs show that all of TLP's settings are applied correctly after resume, so this is not an issue with TLP.

I had a very similar symptom on my ThinkPad T450s. RC6 was only possible when started on AC power. After the first suspend/resume on battery or boot on battery no RC6 possible. Problem was solved by changing the SSD from Liteon to Samsung 850 EVO (tip from a forum).

You may try that or try a newer kernel (didn't help for me) or file a kernel bug. Good luck.

Further reading on the subject: http://mjg59.dreamwidth.org/34868.html

closed 2017-04-15 by linrunner

linrunner, 2017-04-15:
ps. I recommend to remove these boot options

    link_power_management_policy=min_power 
    snd_hda_intel.power_save=2 
    snd_hda_intel.power_save_controller=1

TLP takes care of the corresponding settings already.

Yutsa, 2017-04-15:
Thanks for all the info.

I thought it was maybe not related to TLP but didn't know for sure.

I'll try to compile a kernel and apply the patches from your link.

And I'll remove those options then.