linrunner/TLP #263: tp_smapi lower threshold broken on XX20 series laptops

smclt30p, 2017-04-19:
The Renesas H8S firmware on the XX20 series ThinkPads uses different instructions to set and store battery lower charge threshold info. See: https://github.com/evgeni/tp_smapi/blob/master/README#L183-L184

linrunner, 2017-04-20:
Hi,

install acpi-call(-dkms) and TLP auto-magically uses it to write charge thresholds.

See the [FAQ](http://linrunner.de/en/tlp/docs/tlp-faq.html#kernmod) too.

closed 2017-04-20 by linrunner

smclt30p, 2017-04-20:
That's neat. Is there any documentation on the ThinkPad ACPI interface for other implementations than TLP?

linrunner, 2017-04-20:
I'm not aware of documentation of the ACPI interface itself. If you find one, don't forget to let me know.

If you just want to use battery features without TLP, see [tpacpi-bat](https://github.com/teleshoes/tpacpi-bat) (comes with TLP).