linrunner/TLP #265: Resuming on battery mode, cpu max freq is 100% (ac mode)

biji, 2017-04-27:
### Symptom data

When resuming on battery mode, cpu max freq is 100% (ac mode). running `tlp resume` manually fixes this

### Expected behavior

cpu max freq set to 50% , i set in battery mode

### Actual behavior

cpu max freq is 100% (ac mode)

### Steps to reproduce the problem

suspend/hibernate while AC plugged, unplug AC, resume system
I'm using Fedora 25

Thanks

tlvince, 2017-04-27:
Sounds like the symptoms described in [TLP: CPU_SCALING_MIN_FREQ_ON_BAT not working after suspend/resume](https://bbs.archlinux.org/viewtopic.php?id=224961).

Does running [pstate-frequency](https://github.com/pyamsoft/pstate-frequency/) after resume work for you?

linrunner, 2017-04-27:
Please show the complete output of

    tlp-stat

1. Before suspend
2. After resume – but _before_ manually invoking `tlp resume`

Could be a SELinux specific problem on F25 --> https://bugzilla.redhat.com/show_bug.cgi?id=1403782#c6

Do you observe AVCs during suspend/resume?

biji, 2017-04-30:
The status:
https://gist.github.com/biji/d52f77a8ff4cb3225f2c73dd2e7c663b

linrunner, 2017-11-23:
The AVCs have been fixed: https://bugzilla.redhat.com/show_bug.cgi?id=1403782 . Is your problem solved?

closed 2018-01-24 by linrunner