linrunner/TLP #267: Bluetooth not working with tlp installed: org.bluez.Error.NotReady

apsdehal, 2017-05-18:
### Symptom data

1. Does the problem occur on battery or AC or both?
Both
2. Attach the *full* output of `tlp-stat` via [Gist](https://gist.github.com/) for *all* cases of 1.
[Gist](https://gist.github.com/apsdehal/89bf97ea0d64586ca6a0a2f80d255646)

### Expected behavior

Bluetooth should start on rfkill unblock and service bluetooth start

### Actual behavior
Starts but not able to discover
Error:

Status of `sudo service bluetooth status`
```
May 17 21:47:21 spook bluetoothd[1043]: Current Time Service could not be registered
May 17 21:47:21 spook bluetoothd[1043]: gatt-time-server: Input/output error (5)
May 17 21:47:21 spook bluetoothd[1043]: Not enough free handles to register service
May 17 21:47:21 spook bluetoothd[1043]: Not enough free handles to register service
May 17 21:47:21 spook bluetoothd[1043]: Sap driver initialization failed.
May 17 21:47:21 spook bluetoothd[1043]: sap-server: Operation not permitted (1)
May 17 21:49:01 spook bluetoothd[1043]: Endpoint registered: sender=:1.35 path=/MediaEndpoint/A2DPSource
May 17 21:49:01 spook bluetoothd[1043]: Endpoint registered: sender=:1.35 path=/MediaEndpoint/A2DPSink
May 17 21:49:21 spook bluetoothd[1043]: Failed to set mode: Failed (0x03)
May 17 21:49:31 spook bluetoothd[1043]: Failed to set mode: Failed (0x03)
```

Status of running scan on on bluetoothctl

`Failed to start discovery: org.bluez.Error.NotReady`

### Steps to reproduce the problem
Installing tlp and restarting with my custom config makes bluetooth useless. Uninstalling tlp and restarting everything works fine again.
I am running mint 18.1 serena on lenovo z50

lsusb output:
```
Bus 001 Device 002: ID 8087:8000 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 005: ID 0bda:b728 Realtek Semiconductor Corp. 
Bus 002 Device 004: ID 5986:055e Acer, Inc 
Bus 002 Device 003: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 002 Device 002: ID 045e:0745 Microsoft Corp. Nano Transceiver v1.0 for Bluetooth
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

linrunner, 2017-05-18:
Try to blacklist the USB bluetooth device

> Bus 002 Device 005 ID **0bda:b72**8 control = auto, autosuspend_delay_ms = 2000 -- Realtek Semiconductor Corp. (**btusb**)

See the [FAQ](http://linrunner.de/en/tlp/docs/tlp-faq.html#input). 

Btw: using CPU_SCALING_MAX_FREQ_ON_BAT with the intel_pstate driver is not effective.  To limit power dissipation use CPU_MAX_**PERF**_ON_BAT – refer to [Configuration](http://linrunner.de/en/tlp/docs/tlp-configuration.html#scaling).

apsdehal, 2017-06-01:
Thanks a lot, that actually worked. I also took into consideration suggestions by you. Thanks a lot for creating this awesome project.

closed 2017-06-01 by apsdehal