linrunner/TLP #270: [drm:r600_ring_test [radeon]] *ERROR* radeon: ring 0 test failed

alicektx, 2017-05-23:
Hello linrunner, 
using TLP 0.9, Mint 18, and kernel 4.10.0-19-generic, on Lenovo Ideapad 300 with hybrid graphics (Intel & Radeon).

I get the following error message when the laptop boots on AC:

May 23 02:59:31 Lenovo-300-17ISK kernel: [   26.336643] [drm:r600_ring_test [radeon]] *ERROR* radeon: ring 0 test failed (scratch(0x850C)=0xCAFEDEAD)
May 23 02:59:31 Lenovo-300-17ISK kernel: [   26.336690] [drm:si_resume [radeon]] *ERROR* si startup failed on resume

That is the default TLP configuration, unchanged. What i further tried out:
1) I changed / enabled RUNTIME_PM_BLACKLIST="03:00.0" 
[where 03:00.0 is according to lspci: 
Display controller: Advanced Micro Devices, Inc. [AMD/ATI] Sun XT [Radeon HD 8670A/8670M/8690M / R5 M330] (rev 83)].
2) Then, in addition to the above, i also further removed 'radeon' from "RUNTIME_PM_DRIVER_BLACKLIST=".
In both cases, same behavior - only when uninstalling TLP, boot & logs appear to be ok again....

However, i noticed that i get no such error messages when I boot on battery instead of AC.

One more note: on older 4.4x kernels, no such errors as well - so i'm kinda puzzled now...is it a matter of 4.10, tlp or possibly a combination of both?
[tlp-log.zip](https://github.com/linrunner/TLP/files/1021502/tlp-log.zip)

I've attached a couple of kern / syslog files in case it's of any help...

All the best & thanks in advance for your reply.

linrunner, 2017-05-23:
Hi,

the most important part is missing: the full output of

    sudo tlp-stat

after AC boot and – for comparison – on BAT too. Thanks.

Via [Gist](http://gist.github.com/) – no attachments please.

alicektx, 2017-05-24:
Hi linrunner, 
here is on battery, right after boot:

--- TLP 0.9 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
TLP_DEFAULT_MODE=AC
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
AHCI_RUNTIME_PM_TIMEOUT=15
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=off
WIFI_PWR_ON_BAT=on
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau"
USB_AUTOSUSPEND=1
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0

+++ System Info
System         = LENOVO Lenovo ideapad 300-17ISK 80QH
BIOS           = D5CN47WW
Release        = Linux Mint 18 Sarah
Kernel         = 4.10.0-19-generic #21~16.04.1-Ubuntu SMP Fri Apr 7 08:20:02 UTC 2017 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-4.10.0-19-generic root=UUID=f582ca2c-2435-4b22-9891-b5305a6b698c ro quiet splash vt.handoff=7
Init system    = systemd v229
Boot mode      = UEFI

+++ TLP Status
State          = enabled
Last run       = 04:28:51 AM,     86 sec(s) ago
Mode           = battery
Power source   = battery

+++ Processor
CPU model      = Intel(R) Core(TM) i5-6200U CPU @ 2.30GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors = performance powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   400000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  2800000 [kHz]
[... 335 lines cut ...]
Bus 001 Device 004 ID 0cf3:e360 control = auto, autosuspend_delay_ms =  2000 -- Atheros Communications, Inc.  (btusb)
Bus 001 Device 003 ID 0bda:0129 control = auto, autosuspend_delay_ms =  2000 -- Realtek Semiconductor Corp. RTS5129 Card Reader Controller (rtsx_usb)
Bus 001 Device 002 ID 04f2:b50e control = auto, autosuspend_delay_ms =  2000 -- Chicony Electronics Co., Ltd  (uvcvideo)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)

+++ Battery Status
/sys/class/power_supply/BAT0/manufacturer                   = SANYO
/sys/class/power_supply/BAT0/model_name                     = L15S4E01
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/energy_full_design             =  39670 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  35530 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  35530 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Full

Charge                                                      =  100.0 [%]
Capacity                                                    =   89.6 [%]

+++ Suggestions
* Install smartmontools for disk drive health info
=====================

I've further tested with "radeon.nopm=0" parameter passed to grub, in which case, the warnings that appear with TLP on AC disappear (however, at the penalty of the battery lasting afterwards for at least 40% less time than before). So maybe it's also a 4.10x kernel version problem and not a TLP issue per se? 

Thanks in advance for any further help / assistance.

linrunner, 2017-05-24:
The great majority of issues with TLP are in fact bugs in the kernel driver's pm features.

(1) RUNTIME_PM_BLACKLIST="03:00.0" is not configured currently. But this is not important, because driver blacklisting via RUNTIME_PM_DRIVER_BLACKLIST="radeon ..." actually works as the AC output shows:

> sys/bus/pci/devices/0000:03:00.0/power/control = **auto** (0x038000, Display controller, radeon)

"auto" is the driver default and shows that TLP didn't touch the setting. All other devices are "on".

(2) TLP uses DPM too

> +++ Radeon Graphics
> /sys/class/drm/card1/device/power_dpm_state = performance
> /sys/class/drm/card1/device/power_dpm_force_performance_level = off

and this seems to be the root of your error message. You may remove your boot option and try either to disable DPM on AC by commenting

    #RADEON_DPM_STATE_ON_AC=performance

or try the BAT setting

    RADEON_DPM_STATE_ON_AC=battery

alicektx, 2017-05-25:
Hi linrunner,
deeply thanks for all of your help & support, i really appreciate it, and also sorry for being somewhat late in my reply - as i had to further google quite a bit to get an idea of why / when those specific errors pop up.

None of the TLP tweaks above seem to remove the errors unfortunately - however, i'm convinced 110% this behavior is merely 'triggered' by TLP, and the underlying problem certainly lies very deep elsewhere. By far the most interesting & very similar bug report / thread that i stumbled upon was in Arch's forums here (with kernel 4.8x, yet the very same radeon card...):
[https://bbs.archlinux.org/viewtopic.php?id=218909](url)
Thereby i'm closing this...

All the best & thanks again for all!

closed 2017-05-25 by alicektx

HorselessHorseman, 2017-12-08:
Hi. I've encountered the same problem with the same laptop on ubuntu 16.04. Fortunately, "RADEON_DPM_STATE_ON_AC=battery" seems to solve it.

cj-sv, 2018-03-11:
This has long been closed but I had the same issue. Commenting out the following worked for me.

```
#RADEON_DPM_STATE_ON_AC=performance
#RADEON_DPM_STATE_ON_BAT=battery
```

- Ubuntu 17.10
- Kernel 4.13.0-36-generic
- TLP 1.1