linrunner/TLP #271: Misbehaving invocations of tlp-functions

Edenhofer, 2017-05-26:
### Symptom data

```
systemd[1]: Starting TLP system startup/shutdown...
tlp[1106]: Applying power save settings.../usr/share/tlp/tlp-functions: line 1172: printf: write error: Inv
tlp[1106]: /usr/share/tlp/tlp-functions: line 1172: printf: write error: Invalid argument
tlp[1106]: /usr/share/tlp/tlp-functions: line 1172: printf: write error: Invalid argument
tlp[1106]: /usr/share/tlp/tlp-functions: line 1172: printf: write error: Invalid argument
tlp[1106]: /usr/share/tlp/tlp-functions: line 1172: printf: write error: Invalid argument
tlp[1106]: /usr/share/tlp/tlp-functions: line 1172: printf: write error: Invalid argument
tlp[1106]: /usr/share/tlp/tlp-functions: line 1172: printf: write error: Invalid argument
tlp[1106]: /usr/share/tlp/tlp-functions: line 1172: printf: write error: Invalid argument
tlp[1106]: done.
tlp[1106]: Setting battery charge thresholds...done.
systemd[1]: Started TLP system startup/shutdown.
```

The problem at least occurs when running on battery. The full [tlp-stat output](https://gist.github.com/Edenhofer/87add799e69fae109f18f06a17eeaef9) does not seem to contain any hints on why this happens.

Just as a side note, I am using Btrfs and therefore set `SATA_LINKPWR_ON_BAT` to `max_perfomance`. Furthermore I have tlp version 1.0 installed and use a custom linux 4.11.2 with some kernel self protection project patches.

### Expected behavior

Besides not expecting tlp to print any error, I would at least expect it to properly display them if they occur.

### Steps to reproduce the problem

Start `tlp.service` or boot your system having the service enabled.

linrunner, 2017-06-01:
Hi,

your XPS 13 has the NVMe SSD?

In your tlp-stat output I see 8 SATA links 

    /sys/class/scsi_host/host[0-7]/link_power_management_policy

I think the 7 errors in your log indicate that the corresponding links are not writeable. 

I suggest your try to write them by hand with (as root):

    echo max_performance > /sys/class/scsi_host/host0/link_power_management_policy
    echo max_performance > /sys/class/scsi_host/host1/link_power_management_policy
    ...
    echo max_performance > /sys/class/scsi_host/host7/link_power_management_policy

You may then blacklist the offending scsi hosts via 

    SATA_LINKPWR_BLACKLIST="host1 host2 ..." 

I'll make the write routine more robust with the next release.

Edenhofer, 2017-06-01:
Indeed, I am using an XPS 13 with an NVMe SSD. Disabling the options does  <i>"resolve"</i> the errors.

Thanks for the help.

linrunner, 2017-06-01:
I'll keep this open until the routine is changed.

linrunner, 2017-11-29:
Fixed.

linrunner, 2018-01-24:
Released.

closed 2018-01-24 by linrunner