linrunner/TLP #273: USB issues when using tlp, resulting in slow boot.

PhilJd, 2017-06-01:
First of all thank you for developing tlp :)

### Symptom data

1. Does the problem occur on battery or AC or both?
Both.
2. Attach the *full* output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1.
[gist containing tlp-stat and dmesg output for tlp enabled/disabled](https://gist.github.com/PhilJd/94476889225f903902c5d47a4f239e97)

### Expected behavior
No error occurs.

### Actual behavior
- Getting a `device not accepting address x, error -62`.
- Boot time increases, ~20 seconds
- As my keyboard and touchpad are connected via USB (internally), I can't do anything for another 20 seconds.

### Steps to reproduce the problem
I enabled tlp (`systemctl enable tlp.service). The error and delay occurred on each startup when it was enabled and on never when tlp was disabled.

Please let me know if I forgot something important to add!

linrunner, 2017-06-01:
Hi,

i think this is a kernel issue where activating power saving causes the driver to malfunction.

I'd start with disabling USB autosuspend completely and next try to disable Runtime PM for USB controllers. See [Troubleshooting](http://linrunner.de/en/tlp/docs/tlp-troubleshooting.html) for a strategy to isolate the offending feature.

PhilJd, 2017-06-01:
Thanks for the quick reply! I won't be able to do this in the next few days, but I'll try to come to this as soon as possible.

linrunner, 2017-11-23:
No response --> closed.

closed 2017-11-23 by linrunner