linrunner/TLP #274: Documentation infelicities

ACinnamonUserInUK, 2017-06-03:
It seems to me that it would be good to alter your documentation, both online and in the `tlp` configuration file, on the following two fronts.

(1) The configuration file says

`You *must* disable your distribution's governor settings or conflicts will occur.`

But, how? It would be good to indicate how. If the method varies significantly among distributions, then the file could at least report that fact.

(2) The same file has

```
# Kernel NMI Watchdog:
#   0=disable (default, saves power), 1=enable (for kernel debugging only)
```

Yet, the NMI Watchdog can be used, also, by the `watchdog` utility, which can be used for purposes other than kernel debugging. (One such purpose: make autofs/systemd not hang on shutdown.)

linrunner, 2017-06-04:
Thanks for your remarks – and bringing"infelicities" to my english vocabulary :-)

(1) I think you must have missed the following paragraph and link in the documentation 

> Important: to use this setting you must disable your distribution's governor settings or conflicts will occur. See the [TLP FAQ](http://linrunner.de/en/tlp/docs/tlp-faq.html#scaling).

I just added a small section about other distros there. As for the config file, i think "your distribution" already implies that it depends on the distribution.

And being at it, i added some more text to even more discourage average users from changing the default (docs and [config file](https://github.com/linrunner/TLP/commit/496b8e64fcd9f8734cbf8122bb765998d2f20d76)). 

(2) I try to specify only the original purposes in the config file. Yours is a workaround. Would be more qualified for the FAQ if more people are affected.

ACinnamonUserInUK, 2017-06-04:
Thank you for your reply.

On the distribution's power settings: perhaps we talk past each other. I did see - indeed I quoted - the passage that you say I missed. Still, I think that `http://linrunner.de/en/tlp/docs/tlp-faq.html#scaling` now provides good advice, although you might add that the widely-available program `cpufrequtils` is another way to change scaling settings.

As to 2: you may be right, though I think that a fair amount of people are affected. At the moment I can't find a biggest of the bug reports on the systemd bugtracker but the problem is made somewhat visible by [a websearch](https://www.google.co.uk/#q=github+systemd+shutdown+hang+watchdog).

closed 2017-06-04 by ACinnamonUserInUK