linrunner/TLP #277: ake-on-lan status for device enp1s0

joksik, 2017-06-22:
hi i wonder that i can in tlp turn on option for "Wake-on-lan status for device enp1s0" ? because when i run PowerTop i have this error that is set to BAD

ps when i change to good i have this messages:

>> ethtool -s enp1s0 wol d;

i read powertoop faq but there is only info about MAX_LOST_WORK_SECS_ON_BAT i cant find info about wake-on-lan

linrunner, 2017-06-23:
The setting you're looking for is documented [here](http://linrunner.de/en/tlp/docs/tlp-configuration.html#wol) and it's enabled by default.

Please show the following outputs

    tlp-stat -s
    grep '.*' /sys/class/net/*/device/class
    find /sys/class/net/*/device/ieee80211 -maxdepth 0 -type d

joksik, 2017-06-23:
Is ethtool package needed to Wake On LAN work in tlp???because after i install 
packman -S ethtool 
i dont have this problem about wake on lan reported in powertop. 

if yes ethool did not install with tlp automatically

**tlp-stat -s**

```
+++ System Info
System         = LENOVO Lenovo ideapad 310-15ISK 80SM
BIOS           = 0XCN42WW
Kernel         = 4.12.0-rc6-g41f1830f5a7a #1 SMP PREEMPT Tue Jun 20 10:28:14 CEST 2017 x86_64
/proc/cmdline  = BOOT_IMAGE=/vmlinuz-linux-git root=/dev/sda3 rw raid=noautodetect acpi_osi=Linux pcie_aspm=force quiet elevator=noop i915.enable_guc_loading=1 i915.enable_guc_submission=1 intel_pstate=skylake_hwp i915.enable_psr=1 i915.disable_power_well=0 i915.semaphores=1 i915.enable_rc6=7
Init system    = systemd 
Boot mode      = UEFI

+++ TLP Status
State          = enabled
Last run       = 15:58:08,   1122 sec(s) ago
Mode           = battery
Power source   = battery

Notice: systemd-rfkill.service is not masked -- invoke "systemctl mask systemd-rfkill.service" to correct this!
```

**ls -l /sys/class/net/**
```
razem 0
lrwxrwxrwx 1 root root 0 06-23 15:58 enp1s0 -> ../../devices/pci0000:00/0000:00:1c.0/0000:01:00.0/net/enp1s0
lrwxrwxrwx 1 root root 0 06-23 15:58 lo -> ../../devices/virtual/net/lo
lrwxrwxrwx 1 root root 0 06-23 15:58 wlp2s0 -> ../../devices/pci0000:00/0000:00:1c.5/0000:02:00.0/net/wlp2s0
```

**grep '.*' /sys/class/net/*/device/class**
```
/sys/class/net/enp1s0/device/class:0x020000
/sys/class/net/wlp2s0/device/class:0x028000
```

**find /sys/class/net/*/device/ieee80211 -maxdepth 0 -type d**
`/sys/class/net/wlp2s0/device/ieee80211`

linrunner, 2017-06-24:
In fact the Arch [package](https://www.archlinux.org/packages/community/any/tlp/) has **ethtool** as optional dependency.

closed 2017-06-24 by linrunner