linrunner/TLP #278: Significant loss of energy saving (~2x!) after upgrade v0.9 -> v1.0

ssergiienko, 2017-06-26:
### Symptom data
Usual energy consumption in the idle was 10k-11k mW on tlp-0.9, _lowest_ value was:
```
# tlp-stat -b
--- TLP 0.9 --------------------------------------------

+++ Battery Status
/sys/class/power_supply/BAT0/manufacturer                   = ASUSTeK
/sys/class/power_supply/BAT0/model_name                     = N551-52
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/energy_full_design             =  56160 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  51872 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  50533 [mWh]
/sys/class/power_supply/BAT0/power_now                      =   9860 [mW]
/sys/class/power_supply/BAT0/status                         = Discharging

Charge                                                      =   97.4 [%]
Capacity                                                    =   92.4 [%]
```
But after upgrading to 1.0 it's *constantly* higher, around 16.5k - 19k (16k is lowest):
```
# tlp-stat -b
--- TLP 1.0 --------------------------------------------

+++ Battery Status
/sys/class/power_supply/BAT0/manufacturer                   = ASUSTeK
/sys/class/power_supply/BAT0/model_name                     = N551-52
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/energy_full_design             =  56160 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  51872 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  49831 [mWh]
/sys/class/power_supply/BAT0/power_now                      =  18381 [mW]
/sys/class/power_supply/BAT0/status                         = Discharging

Charge                                                      =   96.1 [%]
```

1. Does the problem occur on battery or AC or both?
BAT
2. Output of `tlp-stat`:
tlp_0.9_AC_origin_settings:
https://gist.github.com/ssergiienko/47c90f031cf8cc388fe253efadee9269
tlp_0.9_BAT_origin_settings :
https://gist.github.com/ssergiienko/52493162273cd3a04e7a0c1bb20bb5d8
 tlp_1.0_AC_updated_settings :
https://gist.github.com/ssergiienko/fc438c707ed0efc7fef12870c669b748
tlp_1.0_BAT_updated_settings :
https://gist.github.com/ssergiienko/c077030858586866d149bbb0dfae54ea

### Steps to reproduce the problem
Just upgrade tlp from 0.9 to 1.0 and restart services.
Reboot doesn't help to fix the issue.
Using config with old settings from tlp 0.9 doesn't change anything.

My guess that this is due to this setting:
```diff
@@ -176,33 +188,39 @@
 RUNTIME_PM_ON_AC=on
 RUNTIME_PM_ON_BAT=auto
 
-# Runtime PM for *all* PCI(e) bus devices, except blacklisted ones:
-#   0=disable, 1=enable
-RUNTIME_PM_ALL=1
-
 # Exclude PCI(e) device adresses the following list from Runtime PM
 # (separate with spaces). Use lspci to get the adresses (1st column).
```

ssergiienko, 2017-06-26:
Just for convenience and syntax highlighting, I leave here diff between my settings from v0.9 and settings that I've change during upgrade to v1.0:
```diff
--- ./tlp_0.9   2017-06-03 12:36:05.746211958 +0300
+++ ./tlp_1.0_merged    2017-06-03 12:35:02.322023993 +0300
@@ -1,5 +1,5 @@
 # ------------------------------------------------------------------------------
-# tlp - Parameters for power save
+# tlp - Parameters for power saving
 # See full explanation: http://linrunner.de/en/tlp/docs/tlp-configuration.html
 
 # Hint: some features are disabled by default, remove the leading # to enable
@@ -12,6 +12,10 @@
 # Concerns some desktop and embedded hardware only.
 TLP_DEFAULT_MODE=AC
 
+# Operation mode select: 0=depend on power source, 1=always use TLP_DEFAULT_MODE
+# Hint: use in conjunction with TLP_DEFAULT_MODE=BAT for BAT settings on AC
+TLP_PERSISTENT_DEFAULT=0
+
 # Seconds laptop mode has to wait after the disk goes idle before doing a sync.
 # Non-zero value enables, zero disables laptop mode.
 DISK_IDLE_SECS_ON_AC=0
@@ -28,7 +32,7 @@
 # Intel Core i processor with intel_pstate driver:
 #   powersave(*), performance
 # Older hardware with acpi-cpufreq driver:
-#   ondemand(*), powersave, performance, conservative
+#   ondemand(*), powersave, performance, conservative, schedutil
 # (*) is recommended.
 # Hint: use tlp-stat -p to show the active driver and available governors.
 # Important:
@@ -46,6 +50,13 @@
 #CPU_SCALING_MIN_FREQ_ON_BAT=0
 #CPU_SCALING_MAX_FREQ_ON_BAT=0
 
+# Set energy performance hints (HWP) for Intel P-state governor:
+#   default, performance, balance_performance, balance_power, power
+# Values are given in order of increasing power saving.
+# Note: Intel Skylake or newer CPU and Kernel >= 4.10 required.
+CPU_HWP_ON_AC=balance_performance
+CPU_HWP_ON_BAT=balance_power
+
 # Set Intel P-state performance: 0..100 (%)
 # Limit the max/min P-state to control the power dissipation of the CPU.
 # Values are stated as a percentage of the available performance.
@@ -156,18 +167,19 @@
 WOL_DISABLE=Y
 
 # Enable audio power saving for Intel HDA, AC97 devices (timeout in secs).
-# A value of 0 disables, >=1 enables power save.
+# A value of 0 disables, >=1 enables power saving.
 SOUND_POWER_SAVE_ON_AC=0
 SOUND_POWER_SAVE_ON_BAT=1
 
 # Disable controller too (HDA only): Y/N
 SOUND_POWER_SAVE_CONTROLLER=Y
 
-# Set to 1 to power off optical drive in UltraBay/MediaBay when running on
-# battery. A value of 0 disables this feature (Default).
+# Power off optical drive in UltraBay/MediaBay: 0=disable, 1=enable.
 # Drive can be powered on again by releasing (and reinserting) the eject lever
 # or by pressing the disc eject button on newer models.
 # Note: an UltraBay/MediaBay hard disk is never powered off.
+#my 1 1
+BAY_POWEROFF_ON_AC=1
 BAY_POWEROFF_ON_BAT=1
 # Optical drive device to power off (default sr0).
 BAY_DEVICE="sr0"
@@ -176,33 +188,39 @@
 RUNTIME_PM_ON_AC=on
 RUNTIME_PM_ON_BAT=auto
 
-# Runtime PM for *all* PCI(e) bus devices, except blacklisted ones:
-#   0=disable, 1=enable
-RUNTIME_PM_ALL=1
-
 # Exclude PCI(e) device adresses the following list from Runtime PM
 # (separate with spaces). Use lspci to get the adresses (1st column).
 #RUNTIME_PM_BLACKLIST="bb:dd.f 11:22.3 44:55.6"
 
-# Exclude PCI(e) devices assigned to the listed drivers from Runtime PM
-# (should prevent accidential power on of hybrid graphics' discrete part).
-# Default is "radeon nouveau"; use "" to disable the feature completely.
+# Exclude PCI(e) devices assigned to the listed drivers from Runtime PM.
+# Default when unconfigured is "amdgpu nouveau nvidia radeon" which
+# prevents accidential power-on of dGPU in hybrid graphics setups.
+# Use "" to disable the feature completely.
 # Separate multiple drivers with spaces.
-RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau"
+RUNTIME_PM_DRIVER_BLACKLIST="amdgpu nouveau nvidia radeon"
 
 # Set to 0 to disable, 1 to enable USB autosuspend feature.
 USB_AUTOSUSPEND=1
 
 # Exclude listed devices from USB autosuspend (separate with spaces).
 # Use lsusb to get the ids.
-# Note: input devices (usbhid) are excluded automatically (see below)
+# Note: input devices (usbhid) are excluded automatically
 #USB_BLACKLIST="1111:2222 3333:4444"
 
-# WWAN devices are excluded from USB autosuspend: 0=do not exclude / 1=exclude
+# Bluetooth devices are excluded from USB autosuspend:
+#   0=do not exclude, 1=exclude
+USB_BLACKLIST_BTUSB=0
+
+# Phone devices are excluded from USB autosuspend:
+#   0=do not exclude, 1=exclude (enable charging)
+USB_BLACKLIST_PHONE=0
+
+# WWAN devices are excluded from USB autosuspend:
+#   0=do not exclude, 1=exclude
 USB_BLACKLIST_WWAN=1
 
 # Include listed devices into USB autosuspend even if already excluded
-# by the driver or WWAN blacklists above (separate with spaces).
+# by the blacklists above (separate with spaces).
 # Use lsusb to get the ids.
 #USB_WHITELIST="1111:2222 3333:4444"
 
@@ -253,6 +271,9 @@
 #START_CHARGE_THRESH_BAT1=75
 #STOP_CHARGE_THRESH_BAT1=80
 
+# Restore charge thresholds when AC is unplugged: 0=disable, 1=enable
+#RESTORE_THRESHOLDS_ON_BAT=1
+
 # ------------------------------------------------------------------------------
 # tlp-rdw - Parameters for the radio device wizard
 # Possible devices: bluetooth, wifi, wwan
```

linrunner, 2017-06-27:
Hi,

thanks for your report.

RUNTIME_PM_ALL is not the cause but you're right about the Nvidia Optimus dGPU. 

Writing **power/control** blocks switching the dGPU power on and off using Bumblebee (see https://github.com/linrunner/TLP/issues/244), so I had to change the behaviour in 1.0. By default all dGPU's are exempted (blacklisted).

I see two workarounds:

1. Disable dGPU blacklisting with

    RUNTIME_PM_DRIVER_BLACKLIST=""

Note: do not comment the line, the empty string is the trick – see [TLP Configuration](http://linrunner.de/en/tlp/docs/tlp-configuration.html#runtimepm).

2. Use a tool like Bumblebee or PRIME to explicitly disable the dGPU upon boot.

Before you ask: I won't answer _any_ questions about those switching tools

ssergiienko, 2017-07-11:
Hi, thanks for an answer, both workarounds are appropriate for me.

> so I had to change the behavior in 1.0

Maybe it would be nice to put such details(or link to comment) in release notes so more people would aware what exactly has changed and how to fix their setups, but it's up to you. You can close this task as resolved, thanks!

linrunner, 2017-07-12:
The release notes actually contain the change:

> Blacklist all dGPU's by default and use kernel default for Runtime PM

The changelog as well:

>  Graphics:
>      ...
>      - Exclude Nvidia dGPU from runtime power management even when no driver
>        is associated (improve compatibility with Bumblebee)

It was way beyond the powers of my imagination to foresee your specific case e.g. that "auto" promptly disables the dGPU. I don't own any hybrid graphics laptop – and stubbornly refuse to do ;-). So I'm very grateful for user reports.

I'll add your case to the FAQ.

linrunner, 2018-01-25:
FAQ changed.

closed 2018-01-25 by linrunner