linrunner/TLP #280: Blacklist USB printers from autosuspending

rimas-kudelis, 2017-07-04:
Coming from #259 (which I would gladly reopen instead, but can't), I think it would be reasonable to allow blacklisting all USB printers automatically, same way WWAN and HID devices are currently blacklisted. Considering that TLP mainly targets portably-used laptops, I think it's safe to assume that if a printer is connected, the user wants to print something. At the very least, the timeout for automatic suspend could be increased for printers.

linrunner, 2017-07-05:
Hi, 

the difficulty is: how to determine that a usb device is a printer? Before you suggest to look for attached drivers: it won't work reliable because udev initializes subdevices asynchronously. 

Implementing the feature for phones was quite a mess (see the [code](https://github.com/linrunner/TLP/blob/master/tlp-usb-udev.in#L127). I'm not keen on a sequel.

Feel free to implement your method and send a pull request. I expect you to organize testing with a significant hardware sample too.

rimas-kudelis, 2017-07-05:
You are blacklisting HID devices by 03 device class. The device class for printers is 07, can't you just blacklist by that?

commit 2540b8a5 (2017-08-08): Allow blacklisting printers from USB autosuspend. Fixes #280.
closed 2017-08-08 by linrunner