linrunner/TLP #281: Implement DEVICES_TO_DISABLE/ENABLE_ON_RESTART options

drws, 2017-07-04:
Please consider implementing additional DEVICES_TO_DISABLE_ON_RESTART and DEVICES_TO_ENABLE_ON_RESTART configuration options which would be effective in case of a reboot. The current two sets of options would still apply for cold (first) boot (_STARTUP) and shutdown (_SHUTDOWN).

linrunner, 2017-07-05:
Hi,

what's your use case for this?

drws, 2017-07-07:
I have my laptop always on and if I reboot, it's usually due to some administrative task and while the reboot takes less than a minute, there is usually no need to change the active radios. But I do turn off my laptop while I travel and such and when in new area, radios are not always welcome. Good examples would be turning the laptop on on an airplane, in a wireless lab or while attending a security conference :)

Another use case would be using it on a pentest machine for similar reasons. Or in case of a hardware failure (cooing failure producing unwanted reboots for example). And then there's always debugging and testing (of crappy wireless cards for example).

linrunner, 2020-06-07:
Reconsidered and rejected because the implementation path – distinction between boot and reboot – is unclear.

closed 2020-06-07 by linrunner

drws, 2021-09-01:
@linrunner Recently I've remembered this proposal and I still think it would be useful. I would kindly ask you to elaborate on your previous answer, specifically why and how the distinction is or would be unclear?