linrunner/TLP #283: Problems with switching power source on MacBook Pro 2017

peterychuang, 2017-07-16:
On 2017 MacBook Pro, there are 3 items in ```/sys/class/power_supply```, and two of them have their types set to "Mains". 

```
/sys/class/power_supply/ADP1/type:Mains
/sys/class/power_supply/BAT0/type:Battery
/sys/class/power_supply/sbs-charger/type:Mains
```

I am not entirely sure what sbs-charger is, but I notice that ```/sys/class/power_supply/sbs-charger/online``` returns ```1``` regardless of wether the power is plugged in. In ```get_sys_power_supply()```, if I understand it correctly, as long as any of the "Mains" sources returns online=1, the function returns that the laptop is on AC. As a result, TLP fails to detect when the power is unplugged, thus the Mode is stuck at AC unless manually switched using ```sudo tlp bat```.

Currently, I use [a workaround](https://github.com/peterychuang/TLP/commit/4389db53244501f21cc81fcc44a1a61c06c99872) to detect switch in power source by checking whether ```/sys/class/power_supply/BAT0/status``` returns "Discharging" instead of wether any "Mains" are online.

linrunner, 2017-07-19:
Hi,

I think a hardcoded exception for "sbs-charger" is appropriate in your case. Checking batteries has it's hazards too. 

Just for the record: could you provide

    tlp-stat -s

please?

peterychuang, 2017-07-19:
>I think a hardcoded exception for "sbs-charger" is appropriate in your case.

I've tried this in [here](https://github.com/linrunner/TLP/blob/master/tlp-functions.in#L329):

```
if [ "$(cat $psrc/online 2> /dev/null)" = "1" ]\
&& [ $psrc != "/sys/class/power_supply/sbs-charger" ]; then
```
This works too, though I'm not sure if that's the way you would go about it.

> Checking batteries has it's hazards too.

I am curious to learn the advantage of checking Mains over checking Battery.

> Just for the record: could you provide
>
> ```
> tlp-stat -s
>```
>
> please?

Here's ```tlp-stat -s``` while on AC:

```
--- TLP 1.0.900 --------------------------------------------

+++ System Info
System         = Apple Inc. 1.0 MacBookPro14,1
BIOS           = MBP141.88Z.0159.B00.1704272110
Release        = "Arch Linux"
Kernel         = 4.13.0-rc1-g486088bc4689 #4 SMP PREEMPT Sun Jul 16 10:10:34 CEST 2017 x86_64
/proc/cmdline  = initrd=\intel-ucode.img initrd=\initramfs-linux-git.img rd.luks.uuid=469a7520-db5c-41d2-830c-83821cc410ba rd.lvm.lv=arch/root rd.lvm.lv=arch/swap resume=UUID=6231a177-fbc2-44b3-820c-04a388185fe0 root=UUID=d2834704-fc9a-4373-ac02-dc5e63db47a1 elevator=none i915.disable_power_well=0 i915.enable_guc_loading=1 i915.enable_guc_submission=1 applespi.fnmode=2 applespi.iso_layout=1 ro quiet
Init system    = systemd 
Boot mode      = UEFI

+++ TLP Status
State          = enabled
Last run       = 08:48:34,   1727 sec(s) ago
Mode           = AC
Power source   = AC
```

Thanks very much.

linrunner, 2017-07-21:
get_sys_power_supply() is a crucial function and it's well tested in it's current form. Battery state detection is much more complicated than AC. Besides a paradigm shift as you propose would mean re-testing on a large number of laptops. Atm I see no reason to take that risk.

linrunner, 2018-01-24:
Released.

closed 2018-01-24 by linrunner