linrunner/TLP #285: Feature Request: pls implement some "Airplane in-seat power mode"

rochelmeyer, 2017-07-28:
Hi all, 
there's a useful feature in lenovos windows-toolbox. "Airplane in-seat power mode" reduces the power consumption in a way that the circuit breaker in airplane-seats, trains etc. does not trip. <https://support.lenovo.com/de/de/solutions/ht062667>
As far as i understand tlp, there is no equivalent feature in tlp yet. 90W-AC-Adaptors frequently draw more current than aircraft-seats or powerbanks can supply (been there - got the t-shirt). Carrying a second ac-adaptor with you for those cases is not a solution. Using Windows is (obviously) not a solution for anything.
Tanks in advance.

linrunner, 2017-07-28:
Hi,

I already saw your thread in thinkpad-forum.de. Afaik there is no kernel interface for this feature and I don't do kernel programming. So it's unlikely the feature can be added in TLP.

rochelmeyer, 2017-07-28:
Thanks or your efforts, anyway!

closed 2017-07-28 by rochelmeyer